ElTangas/jtag2updi #4: getting started

JannyKul, 2018-03-03:
Hey, thanks for providing this firmware.

At the moment I'm struggling to get this working using a mac although I'm sure whatever I'm doing wrong is something blindingly obvious.

These are the steps I've taken:
1. install avrdude via homebrew (it installs version 6.3 successfully)
2. overwrite /usr/local/Cellar/avrdude/6.3/etc/avrdude.conf with modified version provided in jtag2updi
3. put files in jtag2updi/source into new directory called jtag2updi and upload to arduino using the arduino ide
4. connect 10uF capacitor to RST and GND pins on the arduino nano, connect UPDI of tiny1616 to arduinos D6 via 4.7k resistor & connect ground/power to each other
5. run "avrdude -c jtag2updi -P /dev/cu.wchusbserial20 -p t1616"

this is the result I get:
> avrdude: jtagmkII_reset(): timeout/error communicating with programmer (status -1)
> avrdude: initialization failed, rc=-1
>          Double check connections and try again, or use -F to override
>          this check.
> 
> avrdude: jtagmkII_close(): timeout/error communicating with programmer (status -1)
> avrdude: jtagmkII_close(): timeout/error communicating with programmer (status -1)
> 
> avrdude done.  Thank you.

I believe the firmware c++ files have been uploaded correctly to the nano as the arduino ide says "done uploading" with the following report:
> Sketch uses 3262 bytes (10%) of program storage space. Maximum is 30720 bytes.
> Global variables use 528 bytes (25%) of dynamic memory, leaving 1520 bytes for local variables. Maximum is 2048 bytes. 
and the arduinos light is blinking during the upload process.

I've tried a 120Ohm resistor on power and reset instead in case that was the issue to no avail.
The port name is the same as in the Arduino IDE, I've tried using tty.* instead of cu.* to no avail there either.
Just to be thorough I've tried a different arduino nano and restarting the mac also to no avail.

Any ideas for what to try next would be greatly appreciated. Thanks

ElTangas, 2018-03-03:
The first thing to check is, since there are Arduinos with the mega168 and others with mega328, if you have built for the correct chip.

If everything is correct, connect to the Arduino using some serial terminal program (capable of sending hex numbers) at 19200 baud, 8 data bits no parity. Better disconnect the t1616 for this test.
Send the following sequence of hex numbers:
0x1b 0x00 0x00 0x01 0x00 0x00 0x00 0x0e 0x01 0xf3 0x97
The programmer should respond with a string that includes  "JTAGICE mkII".

ElTangas, 2018-03-03:
On the other hand, your errors indicate a problem in communicating with the target t1616.
So carefully check all connections. Then, if the problem persists, go to the file "updi_io.cpp" and change `#define BIT_RATE 250000U`
to
`#define BIT_RATE 200000U`
This slower com rate should help make communication more robust. You can further decrease it if you want, for example to 100000U.

ElTangas, 2018-03-04:
Actually I've now reproduced your error by setting
`#define BIT_RATE 100000U`
This means the source of the problem is my software serial UART implementation (I knew I should have done it in assembly...)
I is sensitive to the compiler you are using and to the compiler options.
Therefore, please copy the compile messages and post them here, so I can analyse them.

Also, if you Arduino is a mega328p, please upload the precompiled binary I have in the build directory and do some tests with that.

JannyKul, 2018-03-04:
You were right, changing to 200000 bit rate fixed the issue - good catch. I assume you're using atmel studio to compile sketches to hex for the new chips? Thanks once again

ElTangas, 2018-03-04:
I develop in Atmel Studio, but the binaries I upload here are actually created using the make.bat file.

This issue was my fault. The datasheet says 225000 baud is the limit, but in testing I wanted to see how far I could go.
On my test chip (a tiny1614) I could go to 250000 but then I forgot to lower to a value within spec...

closed 2018-03-05 by JannyKul