ElTangas/jtag2updi #7: RSP_ILLEGAL_MCU_STATE after setting WDTCFG fuse

lollinopil, 2018-08-23:
When I change the WDTCFG fuse to something different from 0x00 through the command:

> avrdude -c jtag2updi -P com4 -p attiny412 -U fuse0:w:0x00:m -v

It seems that the MCU goes to a lock state. Below is what I receive for any further command:

> C:\Program Files\ConsoleZ.x64.1.18.1.17087>avrdude -c jtag2updi -P com4 -p attiny412 -U flash:w:"d:\projects\christmas_ball\CB_blink\blink_tiny412\bauble_blink_tiny412\Debug\bauble_blink_tiny412.hex":i
> 
> avrdude: jtagmkII_reset(): bad response to reset command: RSP_ILLEGAL_MCU_STATE
> avrdude: initialization failed, rc=-1
>          Double check connections and try again, or use -F to override
>          this check.

In order to avoid this inconvenience, I have to set the WDTCFG fuse back to 0x00, thus disable the WDT.
Is it a known issue?

Thank you for your support

ElTangas, 2018-08-23:
This issue was not known, thanks for reporting.
I will have to look into it.

ElTangas, 2018-08-23:
To help diagnostics, please run this command and post the result:
> avrdude -c jtag2updi -P com4 -p attiny412 -v -v -v -v

I need the level 4 verbosity report to have more detail on the error.

lollinopil, 2018-08-23:
Dear Tangas, below is what I receive:
> C:\Program Files\ConsoleZ.x64.1.18.1.17087>avrdude -c jtag2updi -P com4 -p attiny412 -v -v -v -v
> 
> avrdude: Version 6.3, compiled on Feb 17 2016 at 09:25:53
>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>          Copyright (c) 2007-2014 Joerg Wunsch
> 
>          System wide configuration file is "C:\WinAVR-20100110\bin\avrdude.conf"
> 
>          Using Port                    : com4
>          Using Programmer              : jtag2updi
> avrdude: jtagmkII_open_pdi()
> avrdude: jtagmkII_getsync()
> avrdude: jtagmkII_getsync(): Sending sign-on command:
> avrdude: jtagmkII_send(): sending 1 bytes
> avrdude: Send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97]
> avrdude: jtagmkII_recv():
> avrdude: Recv: . [1b]
> avrdude: Recv: . [00]
> avrdude: Recv: . [00]
> avrdude: Recv: . [1d]
> avrdude: Recv: . [00]
> avrdude: Recv: . [00]
> avrdude: Recv: . [00]
> avrdude: Recv: . [0e]
> avrdude: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]
> avrdude: Recv: < [3c]
> avrdude: Recv: . [7f]
> 
> avrdude: jtagmkII_recv(): Got message seqno 0 (command_sequence == 0)
> avrdude: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]
> 
> Raw message:
> 0x86 0x01 0x01 0x00 0x06 0x01 0x01 0x00 0x06 0x01 0x00 0x00 0x00 0x00 0x00 0x00
> 0x4a 0x54 0x41 0x47 0x49 0x43 0x45 0x20 0x6d 0x6b 0x49 0x49 0x00
> Sign-on succeeded
> 
> JTAG ICE mkII sign-on message:
> Communications protocol version: 1
> M_MCU:
>   boot-loader FW version:        1
>   firmware version:              6.00
>   hardware version:              1
> S_MCU:
>   boot-loader FW version:        1
>   firmware version:              6.00
>   hardware version:              1
> Serial number:                   00:00:00:00:00:00
> Device ID:                       JTAGICE mkII
> avrdude: jtagmkII_getsync(): Using a 298-byte device descriptor
> avrdude: jtagmkII_setparm()
> avrdude: jtagmkII_setparm(): Sending set parameter command (parm 0x03, 1 bytes):
> avrdude: jtagmkII_send(): sending 3 bytes
> avrdude: Send: . [1b] . [01] . [00] . [03] . [00] . [00] . [00] . [0e] . [02] . [03] . [06] $ [24] 1 [31]
> avrdude: jtagmkII_recv():
> avrdude: Recv: . [1b]
> avrdude: Recv: . [01]
> avrdude: Recv: . [00]
> avrdude: Recv: . [01]
> avrdude: Recv: . [00]
> avrdude: Recv: . [00]
> avrdude: Recv: . [00]
> avrdude: Recv: . [0e]
> avrdude: Recv: . [80]
> avrdude: Recv: . [cd]
> avrdude: Recv: . [83]
> 
> avrdude: jtagmkII_recv(): Got message seqno 1 (command_sequence == 1)
> avrdude: Recv: . [80]
> 
[... 298 lines cut ...]
> avrdude: jtagmkII_close(): Sending sign-off command:
> avrdude: jtagmkII_send(): sending 1 bytes
> avrdude: Send: . [1b] . [0a] . [00] . [01] . [00] . [00] . [00] . [0e] . [00] . [a9] . [a0]
> avrdude: jtagmkII_recv():
> avrdude: Recv: . [1b]
> avrdude: Recv: . [0a]
> avrdude: Recv: . [00]
> avrdude: Recv: . [01]
> avrdude: Recv: . [00]
> avrdude: Recv: . [00]
> avrdude: Recv: . [00]
> avrdude: Recv: . [0e]
> avrdude: Recv: . [80]
> avrdude: Recv: . [a1]
> avrdude: Recv: $ [24]
> 
> avrdude: jtagmkII_recv(): Got message seqno 10 (command_sequence == 10)
> avrdude: Recv: . [80]
> 
> Raw message:
> 0x80
> OK
> 
> 
> avrdude done.  Thank you.

ElTangas, 2018-08-23:
Ok, this is the part that matters:

<pre>
Raw message:
0xa5 0xa2
Illegal MCU state</pre>

The raw messages indicates the MCU is in reset, probably because the watchdog timer timed out.
I can modify the program to ignore the reset state, but I'm not sure it will work.
I'll create a test branch, then you can try and report back.

lollinopil, 2018-08-24:
@ElTangas , unfortunately the test code returns an error. I have done the following:

1. Flashed my Arduino Nano with the *.ino sketch found in the source folder
2. As the avedude conf file content did not change, I did not replace it
3. Opened the shell and run the command you suggested

Below the details:

> C:\Program Files\ConsoleZ.x64.1.18.1.17087>avrdude -c jtag2updi -P com4 -p attiny412 -v -v -v -v
> 
> avrdude: Version 6.3, compiled on Feb 17 2016 at 09:25:53
>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>          Copyright (c) 2007-2014 Joerg Wunsch
> 
>          System wide configuration file is "C:\WinAVR-20100110\bin\avrdude.conf"
> 
>          Using Port                    : com4
>          Using Programmer              : jtag2updi
> avrdude: jtagmkII_open_pdi()
> avrdude: jtagmkII_getsync()
> avrdude: jtagmkII_getsync(): Sending sign-on command:
> avrdude: jtagmkII_send(): sending 1 bytes
> avrdude: Send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97]
> avrdude: jtagmkII_recv():
> avrdude: ser_recv(): programmer is not responding
> avrdude: jtagmkII_recv(): Timeout receiving packet
> avrdude: jtagmkII_getsync(): sign-on command: status -1
> avrdude: jtagmkII_getsync(): Sending sign-on command:
> avrdude: jtagmkII_send(): sending 1 bytes
> avrdude: Send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97]
> avrdude: jtagmkII_recv():
> avrdude: Recv: . [1b]
> avrdude: Recv: . [00]
> avrdude: Recv: . [00]
> avrdude: Recv: . [1d]
> avrdude: Recv: . [00]
> avrdude: Recv: . [00]
> avrdude: Recv: . [00]
> avrdude: Recv: . [0e]
> avrdude: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]
> avrdude: Recv: < [3c]
> avrdude: Recv: . [7f]
> 
> avrdude: jtagmkII_recv(): Got message seqno 0 (command_sequence == 0)
> avrdude: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]
> 
> Raw message:
> 0x86 0x01 0x01 0x00 0x06 0x01 0x01 0x00 0x06 0x01 0x00 0x00 0x00 0x00 0x00 0x00
> 0x4a 0x54 0x41 0x47 0x49 0x43 0x45 0x20 0x6d 0x6b 0x49 0x49 0x00
> Sign-on succeeded
> 
> JTAG ICE mkII sign-on message:
> Communications protocol version: 1
> M_MCU:
>   boot-loader FW version:        1
>   firmware version:              6.00
>   hardware version:              1
> S_MCU:
>   boot-loader FW version:        1
>   firmware version:              6.00
>   hardware version:              1
> Serial number:                   00:00:00:00:00:00
> Device ID:                       JTAGICE mkII
> avrdude: jtagmkII_getsync(): Using a 298-byte device descriptor
> avrdude: jtagmkII_setparm()
> avrdude: jtagmkII_setparm(): Sending set parameter command (parm 0x03, 1 bytes):
> avrdude: jtagmkII_send(): sending 3 bytes
> avrdude: Send: . [1b] . [01] . [00] . [03] . [00] . [00] . [00] . [0e] . [02] . [03] . [06] $ [24] 1 [31]
> avrdude: jtagmkII_recv():
> avrdude: Recv: . [1b]
[... 266 lines cut ...]
> 
> avrdude: jtagmkII_reset(): timeout/error communicating with programmer (status -1)
> avrdude: initialization failed, rc=-1
>          Double check connections and try again, or use -F to override
>          this check.
> 
> avrdude: jtagmkII_close()
> avrdude: jtagmkII_close(): Sending GO command:
> avrdude: jtagmkII_send(): sending 1 bytes
> avrdude: Send: . [1b] . [08] . [00] . [01] . [00] . [00] . [00] . [0e] . [08] . [8e] ' [27]
> avrdude: jtagmkII_recv():
> avrdude: ser_recv(): programmer is not responding
> avrdude: jtagmkII_recv(): Timeout receiving packet
> 
> avrdude: jtagmkII_close(): timeout/error communicating with programmer (status -1)
> avrdude: jtagmkII_close(): Sending sign-off command:
> avrdude: jtagmkII_send(): sending 1 bytes
> avrdude: Send: . [1b] . [08] . [00] . [01] . [00] . [00] . [00] . [0e] . [00] . [c6] . [ab]
> avrdude: jtagmkII_recv():
> avrdude: ser_recv(): programmer is not responding
> avrdude: jtagmkII_recv(): Timeout receiving packet
> 
> avrdude: jtagmkII_close(): timeout/error communicating with programmer (status -1)
> 
> avrdude done.  Thank you.

lollinopil, 2018-08-24:
@ElTangas , unfortunately the test code returns an error. I have done the following:

1. Flashed my Arduino Nano with the *.ino sketch found in the source folder
2. As the avedude conf file content did not change, I did not replace it
3. Opened the shell and run the command you suggested

Below the details:

> C:\Program Files\ConsoleZ.x64.1.18.1.17087>avrdude -c jtag2updi -P com4 -p attiny412 -v -v -v -v
>
> avrdude: Version 6.3, compiled on Feb 17 2016 at 09:25:53
>          Copyright (c) 2000-2005 Brian [...]

ElTangas, 2018-08-24:
Yeah, I'm not surprised. I'll have to disable the watchdog timer temporarily during programming, or it keeps resetting the MCU. This may take quite awhile, though.

ElTangas, 2018-08-25:
Ok, I believe this bug is fixed, I did a few tests and it seems to be working now.
Feel free to test it and give some feedback.

closed 2019-07-14 by ElTangas