ElTangas/jtag2updi #11: Embedding inside an Arduino Project

s-ol, 2019-01-30:
This is more of a usage question: is it possible to embed this code inside an existing Arduino Project and still retain the ability to flash a TinyAVR MCU?

Is jtag strictly necessary for the process to work or could you also transmit the data using a custom Serial Protocol?

AProzis, 2019-03-12:
Hi ElTangas. 

I am trying to use an arduino UNO to program my atmega4809. But I am getting stuck on programming.
I configured the atmel studio to use AVRDUDE, but when i try to upload my code, I am getting:
`avrdude.exe : jtagmkII_getsync(): sign-on command: status -1`

This is the configs:
`".....\tools\avr/etc/avrdude.conf" -cjtag2updi -PCOM3 -b115200 -pm4809 -D -Uflash:w:"$(ProjectDir)Release\$(TargetName).hex":i`

Could you help me please?

Thank you!

Balze2, 2021-03-28:
Hello all,

I would like to push this issue up again.

My current project has an ATMega328PB and a Tiny412 on one PCB.
The 238PB does the main part and should also be capable of programming the Tiny.
This works perfect wit your JTAG2UPDI firmware. (Thank you very much for your great work!!!)
I would like to switch (a switch for this is connected to a digital in of the 328PB) between my firmware and your JTAG2UPDI firmware at startup
Sad but true, I don't know how to do this. I'm not familiar with the namespace concept (at least without identifier.!?)
I don't know how to handle your loop() and setup() in this context.
Maybe you (or someone with that knowledge) can guide me.
If someone could combine JTAG2UPDI with the Arduino blink example. I'll love it. <3

Thank you very much in advance

Balze2

Azuzula, 2022-08-29:
Hi, I have same thoughts about combine Arduino as ISP code with jtag2updi code. I have done my own ISP programmer few years ago and want to add UPDI support. Any news about this issue? Thank you.

s-ol, 2022-08-29:
@Balze2 an "anonymous namespace" in C++ is a way to say declare some things locally for a single file only. It's used here just to say "these functions are not called elsewhere"

Azuzula, 2022-08-29:
> @Balze2 an "anonymous namespace" in C++ is a way to say declare some things locally for a single file only. It's used here just to say "these functions are not called elsewhere"

I'm sorry but I have no idea what you talk about. So is or is not possible implement this two codes together? And how can I do that if is it possible? I thought i simply slap the switch on the pcb and then run two different "loops" for ISP or UPDI. Thank you.

s-ol, 2022-08-29:
@Azuzula I replied to someone else, my comment only makes sense if you read that as well.

It's not enough just to switch what firmware to call in loop(), you also have to watch out that they don't cause problems with each other in setup() or by setting interrupts etc. It's a better idea to check the switch at the top of setup() and then run only one project until the arduino is restarted.

Is it possible? Absolutely.

Balze2, 2022-08-29:
> It's a better idea to check the switch at the top of setup() and then run only one project until the arduino is restarted.

That's exactly, what I did. And it works perfectly for me.

Azuzula, 2022-08-29:
> It's not enough just to switch what firmware to call in loop(), you also have to watch out that they don't cause problems with each other in setup() or by setting interrupts etc. It's a better idea [...]

Sure I will but I don't know how to handle empty ino and other code with setup and loop. Should I just run empty loop for UPDI code to run?

Azuzula, 2022-08-29:
> I can post my code here when I'm back home.

Thank you.

Balze2, 2022-08-29:
Keep in mind I'm just a hobbyist!
For sure ElTangas could do better, but this is my approach:
This is working on an ATMega328PB
[main.cpp.txt](https://github.com/ElTangas/jtag2updi/files/9444028/main.cpp.txt)

Balze2, 2022-08-29:
Well, I didn't use the Arduino IDE since years, thus I can't help.
I use "Microsoft Visual Studio Code" (free!) with PlatformIO IDE Extension instead.
Give it a try, it's worth it!

Balze2, 2022-08-29:
Oh, I think I remember...
The Arduino-IDE can't handle a main() function from scratch.
There are solutions I guess but I would recommend VS Code + PlatformIO !!!