ElTangas/jtag2updi #12: ATtiny402 signatures

SimonMerrett, 2019-03-29:
This issue has been partly raised in [this thread](https://www.avrfreaks.net/forum/attiny-202402-was-simulator-issue-now-bad-device-id), independently of jtag2updi and I am now experiencing it combined with an avrdude.conf issue with jtag2updi.

I am a complete novice when it comes to command line operation of avrdude and am slowly getting closer to baremetal, so please go easy if I am making major errors.

Setup: jtag2updi is built using Arduino IDE 1.8.9 and loaded on an Arduino Nano clone with an ATMEGA328P and CH340G USB-Serial converter onboard. 5V VCC and GND are connected between both the Nano and pins 1 and 8 respectively on the ATtiny402. The ATtiny402 is on a SOIC8 to 0.1" header breakout in a breadboard. There's a 4.7k Ohm resistor between Nano pin D6 and pin 6 / PA0 on the ATtiny402. I have installed avrdude v6.3 and have renamed the original avrdude.conf to OLDavrdude.conf and copied the avrdude.conf provided in the jtag2updi repo into the avrdude folder.

I then run this command with avrdude in the command line:
`C:\avrdude>avrdude -c jtag2updi -P com5 -p t402`
and get the response:
```
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.30s

avrdude: Device signature = 0x1e9225 (probably t406)
avrdude: Expected signature for ATtiny402 is 1E 92 23

avrdude done.  Thank you. 
```
Firstly, I believe the avrdude.conf should be using correct device signature for ATtiny402 as **1E 92 27**:
[image]

The second problem is that the device signature is incorrect in the physical chip! May I suggest an additional entry in the jtag2updi version of avrdude.conf along the lines of:
```
#------------------------------------------------------------
# ATtiny402w = workaround to address wrong device signature on some chips
#------------------------------------------------------------

part parent    ".avr8x_tiny"
    id        = "t402w";
    desc      = "ATtiny402w";
    signature = 0x1E 0x92 0x25;

    memory "flash"
        size      = 0x1000;
        offset    = 0x8000;
        page_size = 0x40;
        readsize  = 0x100;
    ;

    memory "eeprom"
        size      = 0x80;
        offset    = 0x1400;
        page_size = 0x20;
        readsize  = 0x100;
    ;
;
```

Thanks for bearing with me. I'm happy to submit a pull request with the edits (datasheet-correct device signature for the 402 and the new 402w entry) but I wanted to check that this makes sense with much more experienced and competent people like @ElTangas, @cbalint13 etc.

SimonMerrett, 2019-03-29:
I tried with the proposed changes to  the avrdude.conf as follows:
```C:\avrdude>avrdude -c jtag2updi -P com5 -p t402w

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.32s

avrdude: Device signature = 0x1e9225 (probably t406)

avrdude done.  Thank you.

C:\avrdude>avrdude -c jtag2updi -P com5 -p t402w -U flash:w:GccApplication1.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.28s

avrdude: Device signature = 0x1e9225 (probably t406)
avrdude: NOTE: Programmer supports page erase for Xmega devices.
         Each page will be erased before programming it, but no chip erase is performed.
         To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude: reading input file "GccApplication1.hex"
avrdude: input file GccApplication1.hex auto detected as Intel Hex
avrdude: writing flash (126 bytes):

Writing | ################################################## | 100% 0.06s

avrdude: 126 bytes of flash written
avrdude: verifying flash memory against GccApplication1.hex:
avrdude: load data flash data from input file GccApplication1.hex:
avrdude: input file GccApplication1.hex auto detected as Intel Hex
avrdude: input file GccApplication1.hex contains 126 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.04s

avrdude: verifying ...
avrdude: 126 bytes of flash verified

avrdude done.  Thank you.
```
And have the nice blinking I hoped for on pin PA6. It's about half the speed I expected from the F_CPU of 8MHz and _delay_ms(250) for both the alternating/toggling HIGH and LOW phases. But it works!

ElTangas, 2019-04-01:
Ok, I agree with your workaround. I will merge your contribution.

SimonMerrett, 2019-04-02:
Thanks @ElTangas . Closing now.

closed 2019-04-02 by SimonMerrett

GrahamPrattWyalong, 2020-05-17:
Hi, I have unfortunately got 25 of these "bad" T402-F chips, I was mucking around with them and have the same problem!

I have updated the UDPI code but still get the same error!

Arduino: 1.8.11 Hourly Build 2019/12/02 05:12 (Mac OS X), Board: "ATtiny412/402/212/202, ATtiny402 (bad signature), 20 MHz, 1.8V, Disabled, Disabled, EEPROM retained, Enabled (default timer), Closer to 5v"
...........
avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: Expected signature for ATtiny402 is 1E 92 27
         Double check chip, or use -F to override this check.
Wrong microcontroller found.  Did you select the right board from the Tools > Board menu?

I am beside myself and about to give up and consider asking RS Australia for a refund on these chips as they do not conform to what is expected, however the avr complaint appears to be telling me I can ignore this error? can you or someone else, advise exactly how to ignore this error (-F seems to be the key)  I am using the Arduino IDE to upload the code and do not know where I would apply this -F that is spoken so highly off?

Any help would be greatly appreciated!

Cheers.

Graham.

SimonMerrett, 2020-05-17:
@GrahamPrattWyalong I found that the fix in the megaTinyCore didn't actually fix this for me either but assumed it didn't affect anyone else and must have been an issue with my toolchain setup. However, now that you have also reported it, there must be some reason the `w` suffix isn't being properly interpreted by avrdude.

My workaround remains to go into the avrdude.conf file and change the signature for the main 402 entry to match what you have in your dud batch. When you use them all up or throw them away and get ones with correct signatures, go back and change the signature back. By then, "we" may have fixed the `w` modifier.

GrahamPrattWyalong, 2020-05-18:
Thanks! Yes, I did exactly that and lo and behold, fixed!

Strange as the w suffix one has the correct value, I did not locate the
selector code and assumed that it always selected the 27 id irrespective of
the intent to select the 25 id!

Cheers for this and I look forward to some interesting projects!

Cheers.

G.

GrahamPrattWyalong, 2020-05-19:
Yes, interest fly enough, I did an install on my mind laptop and the
avrdude that came with that one had an id of 0x23 rather than 0x27 for the
402... naturally I had to change that to 0x25 also! Odd?

G.