ElTangas/jtag2updi #14: Arduino Mega 2560 connection

grogcw, 2019-05-12:
Hi !

I tried to use my Arduino Mega board (one of the firsts rev. I think) and I'm having issues :

I compiled thru Arduino project and uploaded the code, I think this part is fine as the board's led is slightly lit (probably waiting for a usb<->serial signal).

My setup is the following :

Attiny402 -> PA0(UPDI pin) -> 4K9 resistor (close enough) -> Arduino Mega board header pin 6

The Arduino's auto-reset is disabled via a 100uF capacitor across reset and GND.
Attiny's Vdd and GND are connected to 5v and GND on the Arduino Mega board supply should be ok).

I installed avrdude, and using this command in jtag2updi folder :
`avrdude -C avrdude.conf -c jtag2updi -P /dev/cu.usbmodem143411 -p t402
`

... I've got either of these two logs :

```
avrdude: jtagmkII_reset(): timeout/error communicating with programmer (status -1)
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: jtagmkII_close(): timeout/error communicating with programmer (status -1)
avrdude: jtagmkII_close(): timeout/error communicating with programmer (status -1)

avrdude done.  Thank you.
```

... or ...

`avrdude: jtagmkII_getsync(): sign-on command: status -1`

... in loop (so I have to kill it by Ctrl-C.

Maybe the pin I connect the UPDI pin is the wrong one ?

Thanks for taking care and for this awesome project,

Cheers !

ElTangas, 2019-05-15:
Unfortunately, this software is not compatible with the ATMega 2560, the MCU that powers the Arduino Mega.
Probably, the program would be easy to adapt, the problem is that I don't have an Arduino Mega to test it :-(

grogcw, 2019-05-15:
If you need anything from me considering that I have quite an experience with electronics, access to the Mega's board schematics (including the nets and pads of the MCU) along with some MCUs to burn (literally), there's no problem. :)

If you consider it would be an hassle to adapt, you can close this issue.

No hard feelings, I also understand that.

Cheers !

ElTangas, 2019-05-18:
It should not be difficult. Probably only the file updi_io.cpp needs to be modified.
I'll work a bit on it then you can test it (in a week or so, I think).

ElTangas, 2019-05-29:
Ok, the test version is almost ready. I will use the code provided by contributor cbalint13 for the Arduino Mega port.
I'm adding some configuration options which will make the code easier to port to other devices.

ElTangas, 2019-06-03:
I created a test branch: https://github.com/ElTangas/jtag2updi/tree/Experimental-Arduino-Mega-compatible

Note that the binaries on the build folder are for the mega328p, as usual.
To test for the Arduino Mega, you will need to build a proper binary.
You can use either the Arduino IDE, which is automatic if you select the Arduino Mega, or edit and run a makefile.
I've set the UPDI pin to be D3 (TX1/ pin 18 on the Arduino Mega) but you can change it by setting the UPDI_PORT and UPDI_PIN defines. Note that UPDI_PORT must be in the A-G range, higher ports are not supported.

Please test it and if everything is fine I will commit to master.

grogcw, 2019-06-05:
All right, I'll check this tomorrow :)

Thanks for the workaround !

Cheers !

ElTangas, 2019-09-05:
Weel, I'll close this and merge Arduino Mega experimental version to main.

closed 2019-09-05 by ElTangas