ElTangas/jtag2updi #16: Using the jtag2updi chip as a serial bridge

MCUdude, 2019-07-31:
Hi!
Would it be possible to use the jtag2updi chip not only work as a programmer, but also a transparent serial bridge between the USB to serial chip and the target? Preferable also automatic switching, so that we don't need to attach a jumper or something to switch between the two options.

I'm planning to make a development board for the ATmega809/1609/3209/4809 family, and it would be awesome if I could upload new programs and use the serial monitor with the same onboard USB connector.

```
                                          V_prog                 V_target
                                              +-+                     +-+
                                               |                       |
 +----------+          +---------------------+ |                       | +--------------------+
 | PC       |          | Programmer          +-+                       +-+  Target            |
 | avrdude  |          |                     |      +----------+         |                    |
 |       TX +----------+ RX              PD6 +------+   4k7    +---------+ UPDI               |
 |          |          |                     |      +----------+         |                    |
 |       RX +----------+ TX               TX +---------------------------+ RX                 |
 |          |          |                     |                           |                    |
 |          |          |                  RX +---------------------------+ TX                 |
 |          |          |                     |                           |                    |
 |          |          |                     +--+                     +--+                    |
 +----------+          +---------------------+  |                     |  +--------------------+
             JTAGICE MkII                      +-+     UPDI          +-+
             Protocol                          GND     Protocol      GND
```

ElTangas, 2019-08-08:
The mega328P is not very well suited for this. It can be done, sure, but somewhat of a PITA. It would be much easier to use a UPDI series MCU because of their event system and configurable logic, but I haven't come around to port jtag2updi for those chips. Eventually I will.

IMO, the best option would be to use an external switch IC, like the TS5A23157 from TI. It adds a bit of extra cost, but nothing huge.

Regarding auto-switch, I think maybe the DTR line could be used to select between programming/UART bridge modes. This will require a bit of research on how exactly avrdude uses the DTR line.

MCUdude, 2020-12-15:
> Regarding auto-switch, I think maybe the DTR line could be used to select between programming/UART bridge modes. This will require a bit of research on how exactly avrdude uses the DTR line.

I was just thinking of this the other day. I hooked my scope to a CP2104 breakout board with all its handshake pins broken out. Sadly, I couldn't see any toggling when Avrdude connects to the CP2104. Hardware-wise, it would be pretty easy to use a toggle from one of the handshake pins to control some switching logic so the programmer automatically could switch between jtag2updi and USB/serial mode. Do you know if it is possible to toggle a handshake pin on connect without having to modify the Avrdude source code?

```
                                                      V_prog                 V_target
                                                        +-+                     +-+
                                                         |                       |
 +----------+                    +---------------------+ |                       | +--------------------+
 | PC       |                    | Programmer          +-+                       +-+  Target            |
 | avrdude  |                    |                     |      +----------+         |                    |
 |       TX +------------+-------+ RX              PD6 +------+   4k7    +---------+ UPDI               |
 |          |            |       |                     |      +----------+         |                    |
 |       RX +--------+---|-------+ TX                  |                 +---------+ RX                 |
 |          |        |   |       |                     |                 |         |                    |
 |          |        |   |       |                     |                 |    +----+ TX                 |
 |      DTR +----+   |   |       |                     |                 |    |    |                    |
 |          |    |   |   |       |                     |                 |    |    |                    |
 +----------+    |   |   |       +---------------------+                 |    |    +--------------------+
                 |   |   |                                               |    |
                 |   |   |                                               |    |
                 |   |   |                                               |    |
                 |   |   |                              +----------------+----+-----+
                 |   |   +------------------------------+ Switch logic              |
                 |   |                                  | +switch (74LVC1G125 etc.) |
                 |   +----------------------------------+                           |
                 |                                      |                           |
                 +--------------------------------------+                           |
                                                        +---------------------------|
```

jvasileff, 2021-02-15:
I tried using an ATtiny814 as a serial bridge using the event system as suggested, and it worked great. The target was easily able to send data at 2M baud to a CH340E serial chip through the ATtiny. Latency was under 10ns, and forwarding was totally asynchronous and independent of the 814's clock speed.

Separately, I've been able run jtag2updi on an ATtiny1614 just fine.

To get started, I think using a momentary switch and status LED to switch between programming mode and serial proxy mode would be fine, and the ATtiny1614 has plenty of extra pins for that.

So the only remaining item is working out how to have the jtag2updi code support the two modes, which isn't exactly clear to me after spending a few minutes looking a the code.

For reference, forwarding an input pin from the target (`PA6` in this case) to the `TX` pin on the ATtiny (`PB2`) can be done with the three lines below. Forwarding the ATtiny's `RX` pin to the target's `RX` pin could also be done, but I'm not sure that's necessary - the PC's `TX` can simply be connected directly to the target's `RX` directly, and will be ignored by the target while the target is being programmed.

```c
// forward RX_FROM_TARGET (PA6) to TX_TO_CH340E (PB2, EVOUT1)
EVSYS.ASYNCCH0 = EVSYS_ASYNCCH0_PORTA_PIN6_gc; // PA6 -> EVENTSYS CHANNEL 0
EVSYS.ASYNCUSER9 = EVSYS_ASYNCUSER9_ASYNCCH0_gc; // EVENTSYS CHANNEL 0 -> PB2/EVOUT1
PORTMUX.CTRLA |= PORTMUX_EVOUT1_bm; // enable event output to EVOUT1
```

Here's what it looked like on the scope (target's TX in blue, CH340E's RX in yellow, baud rate 2000000):

[image]

jvasileff, 2021-02-17:
I took a look at the avrdude source code, and came to the same conclusion as @MCUdude - a patch would be required to add DTR or RTS signaling. Code for the `arduino` programmer type explicitly sets DTR & RTS, but this is not done for other programmers.

That leaves us with three options I can think of to switch between UPDI and serial modes:

1. Use a hardware button,
2. Use a separate program to send a message using RTS, or
3. Have jtag2updi snoop on all traffic, and switch to updi mode when a valid 'sign-on command' sequence is seen

For number 2, a small python program could be used, such as:

```python
#!/usr/bin/env python3

import time
import serial

ser = serial.Serial('/dev/ttyUSB0')
ser.setRTS(False)
time.sleep(0.1)
ser.setRTS(True)
time.sleep(0.1)
ser.setRTS(False)
time.sleep(0.1)
```

Number 3 may actually be practical, since the sign on sequence is several bytes, and would have to be sent using the correct baud rate to be recognized.

A hybrid approach may also be possible, where a hardware switch (1) is used to enable auto mode switching (3).

WDYT?

I plan on trying all three, but any thoughts or advice would be helpful.

MCUdude, 2021-02-17:
To ensure maximum compatibility and ease of use, I'll probably prefer option 3, where the event system is used to route the signal to a set of spare pins.

I haven't had a look at the Avrdude source code, but I'll assume the beginning of the serial data transfer has a specific "pattern" or bitstream that's possible to detect. One issue might be that you'll at least need to capture two bytes before you can be completely sure that you've receiving jtag2updi commands and not just random serial data.

* We know that the jtag2updi initializing sequence has some known "signature", and the baud rate is 115200.
* When a recognized signature is received we disable the event system (to disable the "standard" RX/TX pins) and "route" the data into the jtag2updi program that's running on the microcontroller. Probably some buffering will be required here because it needs to receive at least two bytes before it knows what to do, and these bytes need to temporarily be stored somewhere so that jtag2updi can do its thing once valid data is received.
* There's probably also possible to automatically switch back to "serial mode" by reading some kind of sign-off package so the programmer is ready to send or receive serial data immedeatly after Avrdude has finished.
* There could also be a switch of some sort to select between "auto" or "manual" serial mode. It's also important that the auto mode is designed in a way that it will be ready for new jtag2updi sign-on commands no matter what happened in the previous session. I'm sure some primitive timing mechanisms can be implemented so that it stays in UPDI mode as long as it's receiving data.

If you succeed, I'm sure this would be a killer product for a marketplace such as [Tindie](https://www.tindie.com). Flexible like the [microUPDI](https://www.google.com/search?client=safari&rls=en&q=microupdi&ie=UTF-8&oe=UTF-8) (which I happen to sell), but where _everything_ is open source, and the only thing that's limiting the user baud rate is the USB to serial adapter itself.

jvasileff, 2021-02-18:
Thanks for the feedback and pointers. I’ll let you know how it goes

jvasileff, 2021-02-18:
I gave it a try, and by simply enabling and disabling the event system when, roughly, `JTAG2::ConnectedTo & 0x02` changes, it works. Except for one *major* problem.

`avrdude` seems to wait until the serial line is idle for 0.25s before doing anything. While data is being received, `avrdude` just waits, not even sending a single byte.

So, a target running the following code can be reprogrammed with auto mode selection:

```
void loop() {
    Serial.println(millis());
    delay(251);
}
```

but if the delay is reduced to `249`, programming always fails. I haven't looked at the `avrdude` code yet, but this might be a show stopper.

jvasileff, 2021-02-18:
Indeed, `avrdude` likes to "drain" the serial port and have it idle for 250ms using the `ser_drain()` function in [ser_posix.c](http://svn.savannah.gnu.org/viewvc/avrdude/tags/RELEASE_6_3/ser_posix.c?view=markup). Adjusting (jtagmkII.c)[http://svn.savannah.gnu.org/viewvc/avrdude/tags/RELEASE_6_3/jtagmkII.c?view=markup] as shown below "fixes" the problem for our auto-switch approach:

```diff
Index: jtagmkII.c
===================================================================
--- jtagmkII.c	(revision 1450)
+++ jtagmkII.c	(working copy)
@@ -1629,7 +1629,11 @@
   /*
    * drain any extraneous input
    */
-  jtagmkII_drain(pgm, 0);
+  // Commenting out. Don't attempt to drain the serial port
+  // before sending the sign-on message to avoid hanging if
+  // the target is continuously sending data (for whatever
+  // reason) prior to receiving the sign-on message.
+  //jtagmkII_drain(pgm, 0);
 
   if (jtagmkII_getsync(pgm, EMULATOR_MODE_PDI) < 0)
     return -1;
```

With this change to `avrdude`, the auto-switch mechanism works very well, even with the target flooding the serial port at a high baud rate. Occasionally, there is a short delay before while (presumably) `avrdude` chokes on buffered data, but never more than a second or two.

But... I'm not sure where that leaves us. Getting a patch to `avrdude` accepted and deployed everywhere seems impractical.

jvasileff, 2021-02-21:
For now, I'm going to turn my attention to a 2 serial port approach, which is probably better overall. So to wrap this up, below is the working PoC patch. I think this is actually pretty close to a "good" patch, but the code would definitely need to be reviewed for behavior during interrupted communication, compatibility with the host timeout feature, etc.

To use it, connect the target's RX pin to the PC's TX pin, and the target's TX to the programmer's PA5 pin, assuming the programmer is an ATtiny1614.

```diff
diff --git c/source/jtag2updi.cpp w/source/jtag2updi.cpp
index 12a0782..a70e9f8 100644
--- c/source/jtag2updi.cpp
+++ w/source/jtag2updi.cpp
@@ -29,6 +29,17 @@ int main(void)
   loop();
 }
 
+void enableSerialBridge(bool enable) {
+    if (enable) {
+        // forward DEST_1_RX (PA5) to PIN_TX (PB2, EVOUT1)
+        EVSYS.ASYNCCH0 = EVSYS_ASYNCCH0_PORTA_PIN5_gc; // PA5 -> CHANNEL 0
+        EVSYS.ASYNCUSER9 = EVSYS_ASYNCUSER9_ASYNCCH0_gc; // CHANNEL 0 -> PB2/EVOUT1
+        PORTMUX.CTRLA |= PORTMUX_EVOUT1_bm; // enable event output to EVOUT1
+    } else {
+        PORTMUX.CTRLA &= ~PORTMUX_EVOUT1_bm; // disable event output to EVOUT1
+    }
+}
+
 /* Internal stuff */
 namespace {
   inline void setup() {
@@ -66,6 +77,9 @@ namespace {
         JTAG2::answer();
         // some commands need to be executed after sending the answer
         JTAG2::delay_exec();
+        if (!(JTAG2::ConnectedTo & 0x2)) {
+          enableSerialBridge(true);
+        }
       #ifndef DISABLE_HOST_TIMEOUT
       } else { // timed out waiting for host communication
         // We timed out waiting for the host to send something
@@ -88,6 +102,7 @@ namespace {
               JTAG2::leave_progmode();
               JTAG2::go();
             }
+            enableSerialBridge(true);
           } else { // send a fail and give them a chance to start talking again
             JTAG2::set_status(JTAG2::RSP_FAILED);
             // send response
@@ -105,6 +120,7 @@ namespace {
   inline void process_command() {
     switch (JTAG2::packet.body[0]) {
       case JTAG2::CMND_GET_SIGN_ON:
+        enableSerialBridge(false);
         JTAG2::sign_on();
         break;
       case JTAG2::CMND_GET_PARAMETER:
```

ckranich, 2021-04-02:
Hi All,

Very interresting discussion! I came down the same road.
Designing a board with a 4809 which should also have a program option via RS-232.
Tried pyupdi and pymcuprog but both did not work well in my tests. also too much at the timing edge possibly.
So decided to go down the "use as many proofen standard components as possible" road..

So my concept currently is
- having a Mega328P _and_ a 4809 both on my board.
328 is spoken to via AVRDude/Arduino IDE as usual.
- DTR line of "Programming" RS-232 Port goes not to Reset of 328 via 0,1u but instead to INT0.
- I will have to add some extension to Jtag2updi (the only place to modify for me)
   329 resets at power on, takes over the serial line (program mode) controlling external XX157 mux
    and timeouts as no host speaks to jtag2updi, 
    releases serial lines (application mode) and enters sleep mode
- DTR line (port openend by IDE for programming or serial monitor) wakes up the 
   328 and either receiving program commands or (switching to application mode again and) entering sleep mode.
- by this I also avoid the 328 bootloader every time    

   I decided against the D11 (used on Nano Every) cause of the simpler to solder 328 and I like the idea of a single 5V supply
   and got rid of all level shifters etc..
   
   Kind Greetings

jvasileff, 2021-04-02:
@ckranich I'm curious, what software do you plan to use on the PC side? One thing that helped push me towards the two-serial-port approach was not having to close my terminal application each time I wanted to flash the chip. The Arduino IDE handles the seamlessly, but I don't want to use the Arduino IDE.

The board I came up with includes a CP2105 (dual uart), an ATtiny 1616 (jtag2updi programmer), and an ATtiny1617 (target):

[image]

ckranich, 2021-04-02:
Hi,

At first it will be the Arduino IDE, but if I make it compatible to existing systems, other IDEs would be possible too.
By using AVRDude as programmer, I think it would be compatible to many settings.
Because space is limited for my project (less than NanoEvery board size) and timetable is limited (I want first samples within 1-2 weeks), I decided to go for 328P as UPDI programmer (I used this already in the past on Arduino compatible designs)
my current setting is 328P (firmware flash via standard Arduino header; possibly smaller physical variant) and a 4809 (using 3 serial ports, where one is used for monitoring). The monitoring port will be shared with 328P.

Currently this is only a very special market, and beeing compatible to Arduino is at the moment just for personal comfort in RAD.
Possibly I will open this later on to specific application engineers (also limited market, maybee 10 .. 20 persons as target), and there the Arduino compatibility might also be a benefit to get them up and running w/o having to explain too much
(time is the most precious ressource in EPUs)

Greetings

Thx for sharing the picture of the board. Is this designed by you?
(At least at a first search I could not find it in the net

jvasileff, 2021-04-02:
Good luck with your project.

> Thx for sharing the picture of the board. Is this designed by you?

Yes. I just got the first board working yesterday.

MCUdude, 2023-08-04:
By using Avrdude 7.2 (or greater), it's now possible to use jtag2updi programmer as a serial bride. This can be achieved by using the DTR/RTS on the USB to serial adapter to control an analog switch, that either routes the Rx and Tx lines to either the jtag2updi microcontroller or to the target.

Pass `-xrtsdtr=low` or `-xrtsdtr=high` to Avrdude to control the state of the DTR/RTS line when programming.

closed 2023-08-04 by MCUdude