ElTangas/jtag2updi #22: JTAG2UPDI on Atmega32U4 (Arduino Leonardo, micro etc)

haweiler, 2019-12-27:
I have some ATtiny202 around to play with - and some 'micro pro' with 32U4 on. So I thought to use ElTangas jtag2updi to program those.

He gave me the tip to have a look at the ATmega4809 with SAMD11 used for USB. So it seemed just to be minor changes. I added those. But I strangely can't compile when changing the board to 'Leonardo' in the Arduino IDE 1.8.10.

[JICE_io.zip](https://github.com/ElTangas/jtag2updi/files/4005969/JICE_io.zip)

ElTangas, 2019-12-29:
Hello, as I mentioned on AVRFreaks, I recommend that you create a fork of jtag2updi to work on the Mega 32U4 port.

haweiler, 2020-02-17:
Hello, I created a fork in jtag2updi, committed the 32U4 port there. Works using Linux right now due to an issue initializing the COM port using Windows.

closed 2020-02-17 by haweiler