ElTangas/jtag2updi #23: JTAG2UPDI on Attiny1614

robsonos, 2020-01-17:
Hi,

It worked the first few times then it stopped working all of the sudden. I didn't touch the fuses so far. Any idea what it may be?

**Command:**
`avrdude -v -v -v -v  -F -c jtag2updi -P COM5 -p attiny1614  -U flash:w:$Attiny1614teste.hex:i`

**Result:**
```

avrdude.exe: Version 6.3-20190619
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Users\robso\Downloads\attiny\attinyMega\jtag2updi\avrdude.conf"

             Using Port                    : COM5
             Using Programmer              : jtag2updi
avrdude.exe: jtagmkII_open_pdi()
avrdude.exe: jtagmkII_getsync()
avrdude.exe: jtagmkII_getsync(): Sending sign-on command: 
avrdude.exe: jtagmkII_send(): sending 1 bytes
avrdude.exe: Send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97] 
avrdude.exe: jtagmkII_recv():
avrdude.exe: Recv: . [1b] 
avrdude.exe: Recv: . [00] 
avrdude.exe: Recv: . [00] 
avrdude.exe: Recv: . [1d] 
avrdude.exe: Recv: . [00] 
avrdude.exe: Recv: . [00] 
avrdude.exe: Recv: . [00] 
avrdude.exe: Recv: . [0e] 
avrdude.exe: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00] 
avrdude.exe: Recv: < [3c] 
avrdude.exe: Recv: . [7f] 

avrdude.exe: jtagmkII_recv(): Got message seqno 0 (command_sequence == 0)
avrdude.exe: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00] 

Raw message:
0x86 0x01 0x01 0x00 0x06 0x01 0x01 0x00 0x06 0x01 0x00 0x00 0x00 0x00 0x00 0x00
0x4a 0x54 0x41 0x47 0x49 0x43 0x45 0x20 0x6d 0x6b 0x49 0x49 0x00 
Sign-on succeeded

JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
avrdude.exe: jtagmkII_getsync(): Using a 298-byte device descriptor
avrdude.exe: jtagmkII_setparm()
avrdude.exe: jtagmkII_setparm(): Sending set parameter command (parm 0x03, 1 bytes): 
avrdude.exe: jtagmkII_send(): sending 3 bytes
avrdude.exe: Send: . [1b] . [01] . [00] . [03] . [00] . [00] . [00] . [0e] . [02] . [03] . [06] $ [24] 1 [31] 
avrdude.exe: jtagmkII_recv():
avrdude.exe: Recv: . [1b] 
avrdude.exe: Recv: . [01] 
avrdude.exe: Recv: . [00] 
avrdude.exe: Recv: . [01] 
avrdude.exe: Recv: . [00] 
avrdude.exe: Recv: . [00] 
avrdude.exe: Recv: . [00] 
avrdude.exe: Recv: . [0e] 
[... 259 lines cut ...]
avrdude.exe: Device signature = 0x666c61
avrdude.exe: Expected signature for ATtiny1614 is 1E 94 22
avrdude.exe: NOTE: Programmer supports page erase for Xmega devices.
             Each page will be erased before programming it, but no chip erase is performed.
             To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude.exe: jtagmkII_close()
avrdude.exe: jtagmkII_close(): Sending GO command: 
avrdude.exe: jtagmkII_send(): sending 1 bytes
avrdude.exe: Send: . [1b] . [08] . [00] . [01] . [00] . [00] . [00] . [0e] . [08] . [8e] ' [27] 
avrdude.exe: jtagmkII_recv():
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: jtagmkII_recv(): Timeout receiving packet

avrdude.exe: jtagmkII_close(): timeout/error communicating with programmer (status -1)
avrdude.exe: jtagmkII_close(): Sending sign-off command: 
avrdude.exe: jtagmkII_send(): sending 1 bytes
avrdude.exe: Send: . [1b] . [08] . [00] . [01] . [00] . [00] . [00] . [0e] . [00] . [c6] . [ab] 
avrdude.exe: jtagmkII_recv():
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: jtagmkII_recv(): Timeout receiving packet

avrdude.exe: jtagmkII_close(): timeout/error communicating with programmer (status -1)

avrdude.exe done.  Thank you.
```

ElTangas, 2020-01-18:
> Any idea what it may be?
Not really. If you didn't change the fuses that's strange, because it seems the tiny is not responding to the programmer.
All I can suggest is, check all connections and power cycle both the programmer and the target chip.
If you have another tiny1614 to test, can you try?

robsonos, 2020-01-20:
I believe I killed it somehow . Unfortunately I don't another tiny to confirm my theory.  I'll get a new one, but it will take at least a month.