ElTangas/jtag2updi #24: UPDI on ATTiny814

haindvn, 2020-01-28:
I have tried this with ATTiny412 and ATTiny814, the 412 works well so far and I can re-program the 412 many times, however, I have problems with ATTiny814.

Regarding 814, I can program the chip once then I can not communicate with the chip anymore,
I run this command try to unlock the chip

avrdude -C UDPI.conf -c jtag2updi -PCOM10 -p t814 -t -F
Then....
avrdude: jtagmkII_reset(): bad response to reset command: RSP_ILLEGAL_MCU_STATE
avrdude: initialization failed, rc=-1
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x656570
avrdude: Expected signature for ATtiny814 is 1E 93 22
avrdude> erase
>>> erase
avrdude: erasing chip
avrdude: jtagmkII_chip_erase(): timeout/error communicating with programmer (status -1)
avrdude>

I always receive the time out communicating with the chip then I can not perform erasing. Is there another way to unlock the chip? Once the chip was programmed, it will be locked even there's no fuse touch?

Thanks.

haindvn, 2020-01-31:
I have tried this on an Uno and got the same issue, I can reflash ATTiny412 easily but ATTiny814, the chip is locked once I programmed the blinky sketch and I can not unlock it. The chip is running normally since I power it and the LED blink as it should (I just simply toggle PA3 pin for blinking). Should I try this on another ATTiny814?
I also tried another version of UDPI as STK2UDPI (yours as well) but it doesn't work because STK500 doesn't recognize new ATTiny series.

ElTangas, 2020-02-02:
Ok. I guess I'll have to get some tiny814 to investigate. Unfortunately I don't have any ATM.

haindvn, 2020-02-03:
Thanks ElTangas for the great tool, do you need more verbose information to investigate? I will get the output tonight.
Currently I have LG8F328, Nano and Uno flashed as UDPI and got this problem with all of them on 814. I think we need to tweek the code in some way, sorry I don't get familiar with UDPI protocol :), I will try to provide information as much as I can if that can help you.

ElTangas, 2020-02-03:
Ok, post the verbose output, that is, add -vvvv to the avrdude command line.

haindvn, 2020-02-04:
Thanks ElTangas, please see the output, hope it helps.

avrdude -C updi.conf -c jtag2updi -PCOM7 -p t814 -e -v -v -v -v

avrdude: Version 6.3, compiled on Feb 17 2016 at 09:25:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "updi.conf"

         Using Port                    : COM7
         Using Programmer              : jtag2updi
avrdude: jtagmkII_open_pdi()
avrdude: jtagmkII_getsync()
avrdude: jtagmkII_getsync(): Sending sign-on command:
avrdude: jtagmkII_send(): sending 1 bytes
avrdude: Send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97]
avrdude: jtagmkII_recv():
avrdude: Recv: . [1b]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [1d]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [0e]
avrdude: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]
avrdude: Recv: < [3c]
avrdude: Recv: . [7f]

avrdude: jtagmkII_recv(): Got message seqno 0 (command_sequence == 0)
avrdude: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]

Raw message:
0x86 0x01 0x01 0x00 0x06 0x01 0x01 0x00 0x06 0x01 0x00 0x00 0x00 0x00 0x00 0x00
0x4a 0x54 0x41 0x47 0x49 0x43 0x45 0x20 0x6d 0x6b 0x49 0x49 0x00
Sign-on succeeded

JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
avrdude: jtagmkII_getsync(): Using a 298-byte device descriptor
avrdude: jtagmkII_setparm()
avrdude: jtagmkII_setparm(): Sending set parameter command (parm 0x03, 1 bytes):
avrdude: jtagmkII_send(): sending 3 bytes
avrdude: Send: . [1b] . [01] . [00] . [03] . [00] . [00] . [00] . [0e] . [02] . [03] . [06] $ [24] 1 [31]
avrdude: jtagmkII_recv():
avrdude: Recv: . [1b]
avrdude: Recv: . [01]
avrdude: Recv: . [00]
avrdude: Recv: . [01]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [0e]
avrdude: Recv: . [80]
avrdude: Recv: . [cd]
avrdude: Recv: . [83]

avrdude: jtagmkII_recv(): Got message seqno 1 (command_sequence == 1)
avrdude: Recv: . [80]
[... 297 lines cut ...]
avrdude: jtagmkII_recv():
avrdude: Recv: . [1b]
avrdude: Recv: . [0a]
avrdude: Recv: . [00]
avrdude: Recv: . [02]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [0e]
avrdude: Recv: . [a5] . [00]
avrdude: Recv: B [42]
avrdude: Recv: A [41]

avrdude: jtagmkII_recv(): Got message seqno 10 (command_sequence == 10)
avrdude: Recv: . [a5] . [00]

Raw message:
0xa5 0x00
Illegal MCU state: Stopped

avrdude: jtagmkII_close(): bad response to sign-off command: RSP_ILLEGAL_MCU_STATE

avrdude done.  Thank you.

Hope this helps :) thanks

ElTangas, 2020-02-04:
> Raw message:
0xa5 0x00
Illegal MCU state: Stopped

State 0x00 shouldn't even be possible!
Ok, I will prepare a debug version to try to get more info.

ElTangas, 2020-02-04:
Here is the test code: https://github.com/ElTangas/jtag2updi/tree/Debug

Please repeat the erase processes and post the verbose output, hopefully it will have more clues.

ElTangas, 2020-02-05:
I've been able to reproduce the error on your first post:
```
avrdude: jtagmkII_recv(): Got message seqno 8 (command_sequence == 8)
avrdude: Recv: . [a5] . [88]

Raw message:
0xa5 0x88
Illegal MCU state

avrdude: jtagmkII_reset(): bad response to reset command: RSP_ILLEGAL_MCU_STATE
avrdude: initialization failed, rc=-1
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x656570
avrdude: Expected signature for ATtiny1614 is 1E 94 22
```

I did this by putting a pull-down on the UPDI link, then removing it. Please check your connections and make sure there is no extra stuff connected to the UPDI link.

haindvn, 2020-02-05:
> Here is the test code: https://github.com/ElTangas/jtag2updi/tree/Debug
> 
> Please repeat the erase processes and post the verbose output, hopefully it will have more clues.

Please find the output of debug version for your debug, hope this helps. I'm using an Uno as UPDI programmer.

```
avrdude -C updi.conf -c jtag2updi -PCOM7 -p t814 -e -vvvv

avrdude: Version 6.3, compiled on Feb 17 2016 at 09:25:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "updi.conf"

         Using Port                    : COM7
         Using Programmer              : jtag2updi
avrdude: jtagmkII_open_pdi()
avrdude: jtagmkII_getsync()
avrdude: jtagmkII_getsync(): Sending sign-on command:
avrdude: jtagmkII_send(): sending 1 bytes
avrdude: Send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97]
avrdude: jtagmkII_recv():
avrdude: Recv: . [1b]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [1d]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [0e]
avrdude: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]
avrdude: Recv: < [3c]
avrdude: Recv: . [7f]

avrdude: jtagmkII_recv(): Got message seqno 0 (command_sequence == 0)
avrdude: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]

Raw message:
0x86 0x01 0x01 0x00 0x06 0x01 0x01 0x00 0x06 0x01 0x00 0x00 0x00 0x00 0x00 0x00
0x4a 0x54 0x41 0x47 0x49 0x43 0x45 0x20 0x6d 0x6b 0x49 0x49 0x00
Sign-on succeeded

JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
avrdude: jtagmkII_getsync(): Using a 298-byte device descriptor
avrdude: jtagmkII_setparm()
avrdude: jtagmkII_setparm(): Sending set parameter command (parm 0x03, 1 bytes):
avrdude: jtagmkII_send(): sending 3 bytes
avrdude: Send: . [1b] . [01] . [00] . [03] . [00] . [00] . [00] . [0e] . [02] . [03] . [06] $ [24] 1 [31]
avrdude: jtagmkII_recv():
avrdude: Recv: . [1b]
avrdude: Recv: . [01]
avrdude: Recv: . [00]
avrdude: Recv: . [01]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [0e]
avrdude: Recv: . [80]
[... 313 lines cut ...]
avrdude: Recv: A [41]

avrdude: jtagmkII_recv(): Got message seqno 10 (command_sequence == 10)
avrdude: Recv: . [a5] . [00]

Raw message:
0xa5 0x00
Illegal MCU state: Stopped

avrdude: jtagmkII_close(): bad response to sign-off command: RSP_ILLEGAL_MCU_STATE

avrdude done.  Thank you.
```

> edit: one more thing, did you change any fuses?

No, I didn't change any fuse. I just tried to upload the simple blinky sketch and it works for the very first time (the blinky sketch is still running when I provide power to the chip via programmer / LED toggle on/off via PA3 pin).Tthen the chip is completely locked and I can not program or erase it anymore. I'm using this setup to program 412 again without any problem (tried 10 times with Tiny412 and 100% works). 
My setup: 
- A resistor 4.7K between Uno D6 and Tiny814 pin 10 (PA0) or Tiny412 pin 6 (PA0)
- VCC & GND from Uno
- Cap 10uF between Uno RESET & GND to avoid auto reset.

Should I get another Tiny814 and try it again? I will buy some chips this weekend and give them a try.

Thanks.

ElTangas, 2020-02-05:
That chip is definitely not behaving normally, but there is no need to risk new chips yet.
The part that interests me is:

> avrdude: jtagmkII_recv(): Got message seqno 8 (command_sequence == 8)
> avrdude: Recv: . [a5] . [88]
> 
> Raw message:
> 0xa5 0x88
> Illegal MCU state

That is, the response to the reset command.

Can you please repeat the avrdude -e  several times and see if the response is always the same, i.e. 0xa5 0x88?
If so I will modify the debug version to accept this value, maybe that will allow the chip to be erased, though I have no idea of what is happening.

haindvn, 2020-02-06:
> **That chip is definitely not behaving normally**, but there is no need to risk new chips yet.
> The part that interests me is:
>
>
> That is, the response to the reset command.
>
> Can you please repeat the [...]

> **That chip is definitely not behaving normally**

Maybe you're correct, I have just tried to run avrdude to erase the chip and it works (I success only three times among 45-50 times, I think). I tried this method:
- Prepare the command line avrdude with parameters, but I don't hit Enter
- Plug the Uno into USB port, once the LED 13 is lit, I immediately hit the Enter for avrdude run

It erases the chip successfully 3 times only with these steps, and now I can not replicate again, it's quite hard to get it done.

Please stay tuned, I will get some Tiny814 this weekend and give them a try. I soldered this Tiny814 on breakout board with heat gun + solder paste, maybe that I push too much hot air so the chip is damaged, maybe, just remembered it's nearly burnt because I'm too rush.
I will update you next week, I will try this on another Tiny814 asap. Thanks :)

haindvn, 2020-02-09:
Hi ElTangas,
I bought some Tiny814 and gave a try on a new chip (don't use hot air gun and solder paste this time, manual soldering, it works as expected without any problem, I tried on LGT8F328 as well as Uno, they all work). Very sorry for any inconvenience caused, it seems that my old chip was damaged during soldering on breakout board with too much hot air applied. We can close the issue now and I'm terrible sorry for raising this issue.

closed 2020-02-09 by haindvn

Tigeruno, 2022-12-24:
I'm having the same issue with attiny841, BUT, it has been used quite alot and is on a used DIYBMS board for a test bed.  
Setup is Nano with your program 4.7k D6 and 10uF cap across RST - GND.
Unable to erase.
avrdude -c jtag2updi -p t841 -P COM6 -b 9600 -F -e -U flash....etc flash hex file
avrdude.exe: jtagmkII_initialize(): part ATtiny841 has no PDI interface
avrdude.exe: initialization failed, rc=-1
avrdude.exe: AVR device initialized and ready to accept instructions
avrdude.exe: Device signature = 0x656570
avrdude.exe: Expected signature for ATtiny841 is 1E 93 15

using avrdudess.

voneiden, 2022-12-26:
@Tigeruno ATtiny841 doesn't have UPDI (reads on your log too).  Do you have a typo (814 vs 841)?

Tigeruno, 2022-12-27:
My bad then.  I thought it would work on any attiny.
Tiny841 does program with ISP (6 pin), but that means I have to disconnect all the circuit wiring using that connection.