ElTangas/jtag2updi #27: Doesn't work on AVR-DA

WestfW, 2020-05-06:
The new AVR-DA series (ie AVR128DA48) has a significantly different NVM controller from the Mega0/XTiny chips.  (Sigh.)  Since flash programming is done by writing the appropriate commands to the NVM controller, you need new code for programming the AVR-DA chips.

ElTangas, 2020-05-06:
Yes I know... the implementation is so different that it will require a lot of code to be written from scratch, it might take a while for me to get around to do it :-(

SpenceKonde, 2020-05-07:
I'm working on this - I think I'm pretty close (ie, a few days) so I'll put in a PR if/when I get it working :-) The logistics of the PR are tricky since git can't merge my fork of it because I moved things around so that someone could compile it for Arduino out of the box.

And yeah - on AVR-DA, erasing the page and writing the page are separate operations, sort of required by the way that flash doesn't need to be written page-by-page now. In practice, since AVRdude does a chip erase before uploading flash anyway, this doesn't really matter (though we could easily have it erase the page first anyway)

I'll have to (very) respectfully disagree with WestFW on which implementation is better. Not having to write a whole page at a time, and with the way EEPROM can be written in byte-oriented, erase-while-writing (EEERWR) mode is pretty damned cool. 

I can't say I understand why they bothered with a million commands to erase different numbers of pages or bytes at a time, though.,..

ElTangas, 2020-05-08:
```
In practice, since AVRdude does a chip erase before uploading flash anyway,
this doesn't really matter (though we could easily have it erase the page first anyway)
```

Regarding this, let me tell about a secret feature:
Go to the "AVR8X family common values" section of avrdude.conf and replace
```
 has_updi = yes;
```
 by
```
has_pdi	= yes;
```

Now there is no longer an auto erase, and this was the intended behaviour I wanted when I wrote jtag2updi, because the AVR-0/1 have an erase/write feature that allows each page to be written individually, erasing the whole thing every time is kind of overkill.

But when avrdude was modified to work with the Nano Every and such, this UPDI option was added.

ElTangas, 2020-05-16:
Anyway, I have some DA chips available, so I'll look into this.
Reading memory is already working fine, since it doesn't involve the NVM controller.
But it's limited to 64kB.
So, the first thing to do, is updating the UPDI low level module, to get it to read the entire memory.

SpenceKonde, 2020-05-16:
I've got this working - I'll be submitting new code probably today or
tomorrow.

ElTangas, 2020-05-16:
Ok, but there's no need to rush this IMO, I'm pretty sure we have a year or so before the DA series starts gaining popularity.
After all, the AVR-0/1 have been around for a few years and only now they are becoming popular.

ElTangas, 2020-05-17:
I have the read-only version for AVR-DA almost ready, just ironing out some details and bugs that emerged.
For example, the UPDI "repeat" instruction only allows 255 repeats, but flash pages have 512 bytes, this may need some rewriting of the code.
I've also dumped the entire address space as seen by the UPDI unit, and located the flash memory at address 0x800000 (the real one, not the remapped one), this doesn't seem to be documented in the datasheet.
All the addresses that do not map to I/O or non-volatile memory seem to default to SRAM, which is endlessly aliased throughout the address space. Kinda interesting.

So given this info I can write avrdude.conf definitions for the 128kB AVR-DA chips, the other ones have no datasheets yet. I guess I could get the signatures, etc. for those from Atmel packs definition files.

ElTangas, 2020-05-17:
@WestfW 
```
Oh! Interesting: the NVMCTRL->CTRLA command used to program mega0 (3: erase and write page)
 is not documented as even existing on AVR-DA (which has 0,1,2, 8...0x20, 0x30.)
```

That's ok because avrdude doesn't use that for flash anyway, it always sends an erase page command first, then a write page command.
So jtag2updi doesn't use the ERWP command for flash, just for EEPROM.
The AVR-DA seem to have erase write for EEPROM only with 1 byte granularity, and for fuses which use the same commands, so says the datasheet, at least...

SpenceKonde, 2020-05-17:
Yeah - that option does not exist in the AVR-DA programming paradigm

ElTangas, 2020-05-30:
AVR-DA support added, closing.

closed 2020-05-30 by ElTangas

hoka83, 2024-04-23:
hello does jtag2updi support avr128da28 to be programmed by arduino uno?