ElTangas/jtag2updi #28: Port manipulation issue/question

SpenceKonde, 2020-05-07:
Maybe not an issue, but why do you do this?

`
        /* Enable all UPDI port pull-ups */
        PORT(UPDI_PORT) = 0xFF;
`

`
        /* Enable all LED port pull-ups, except for the LED pin */
        PORT(LED_PORT) = 0xFF - (1 << LED_PIN);
`

Also, as an aside, on XTINY (ahem, tinyAVR 0-series and 1-series), as well as megaAVR 0-series) that doesn't even do what the comment says, as the pullup is controlled by the PINnCTRL register instead...

Definitely not intended behavior:
You have an LED_PORT and LED_PIN - but you never set DDR for it so it never turns on - it looks like you meant to do that here?

`
        /* Enable LED */
        PORT(LED_PORT) |= (1 << LED_PIN);
`

SpenceKonde, 2020-05-07:
As an aside, for proof that it's not necessary to turn on the pullups, consider the fact that you... don't do it on the XTINY parts. 

VPORTx.OUT does **not** emulate the old pullup behavior of classic AVRs, setting an input high this way does not turn on the pullups..

ElTangas, 2020-05-07:
Probably I enabled the pullups to avoid letting the inputs floating, I do this out of habit sometimes but it's not really necessary.

Regarding the pullup activation not working on AVR-0/1, I only found out about this after writing that code and forgot about it.