ElTangas/jtag2updi #30: Does the TEMP register stuff actually save us anything?

SpenceKonde, 2020-05-12:
It seems hard to believe that this could make that much of a difference... 

I didn't see a performance difference using it, thought it made a small difference in the sketch size. Wondering if it's worth even the mess of looking at, since I'm pretty sure the bottle neck isn't shuffling the data to and from the software serial port (and probably actually dealing with said software)

SpenceKonde, 2020-05-12:
But all the XAVR parts have a full complement of 4 registers (plus, we now need to handle 24-bit addresses)....

ElTangas, 2020-05-12:
It was an experiment I did because I didn't like the code avr-gcc was generating. Yes, it's only meant to reduce code size, not relevant for performance.

In fact not much on the UPDI side is relevant for performance provided it can handle the UPDI settings that I use; the current main limitation is that the jtagice mkII protocol only supports a maximum of 115200 baud.

closed 2020-05-17 by SpenceKonde