ElTangas/jtag2updi #34: Why Target Differentiation?

mikrocoder, 2020-05-22:
Hi,

... I think there is a typo in the parts.h file. The ending 9 at the end is missing in the last line. 809 ... 4809.
https://github.com/SpenceKonde/jtag2updi/blob/master/parts.h

best regards

ElTangas, 2020-05-23:
I think you are right, I'll forward this issue to @SpenceKonde, who contributed this file.

mikrocoder, 2020-05-23:
> I think you are right, I'll forward this issue to @SpenceKonde, who contributed this file.

Thanks for forwarding. I would have preferred to write it directly to Spence Fork, but it lacks an issues tab.   :-)

SpenceKonde, 2020-05-23:
Oh, huh - I'll look into why that tab is missing - though I'm in the
process of submitting my changes back to this repo anyway.

SpenceKonde, 2020-05-23:
I mean, the fork on my github - once I get everything sorted out and
submitted here, is supposed to differ *only* in that it the containing
folder stripped out so that you can compile it in Arduino without moving
anything around. So maybe that's why I didn't put an issues tab. Once
things are in the state that they're supposed to be in, since the code in
my repo will be the same as the code here, it really wouldn't be
appropriate to report issues there, because I'd just have to go and report
them here!

ElTangas, 2020-05-27:
This has been corrected, closing.

closed 2020-05-27 by ElTangas