ElTangas/jtag2updi #36: modified avrdude.conf has errors

maartenvanschagen, 2020-06-17:
Hey ElTangas,

It seems like the avrdude.conf provided with this repository has a few issues on my pc (arch linux with Avrdude installed from Pacman).
I get the following error (the config file is called old.conf):
`$ avrdude -C old.conf -c jtag2updi -P /dev/ttyUSB0 -p t402 -U flash:w:test.hex
avrdude: error at old.conf:1115: programmer type jtagice3_updi not found
avrdude: error reading system wide configuration file "old.conf"`
To fix this issue I got rid of any programmer of the jtagice3_updi type (including snap_updi). Now I get the following error:
`$ avrdude -C old.conf -c jtag2updi -P /dev/ttyUSB0 -p t402 -U flash:w:test.hex
avrdude: error at old.conf:15183: syntax error
avrdude: error reading system wide configuration file "old.conf"`
To fix this error I had to remove every decleration of ocd_base.

After these changes, everything runs great.

maartenvanschagen, 2020-06-17:
It doesn't seem like avrdude has a definition of jtagice3_updi at all. I can't find it on the internet, and my systemwide configuration doesn't include it either. Is there something I'm missing?

ElTangas, 2020-06-18:
I've just made minor modifications to the avrdude.conf file from the main repository of avrdude from here: http://svn.savannah.gnu.org/viewvc/avrdude/trunk/avrdude/avrdude.conf.in?revision=1422&view=markup

I didn't just invent the jtagice3_updi programmer, it comes from the upstream source.

maartenvanschagen, 2020-06-19:
Oh, sorry, I never found those files, I thought 6.3 was the newest version of avrdude. I guess I have an old version installed

maartenvanschagen, 2020-06-19:
Oh, I now see you provided that link in the readme too. I didn't see that avrdude also releases the trunk version of their software (when I wrote my earlier post I thought this was a patched variant of avrdude).

To avoid confusion, it would be nice if this was clarified in the readme. Right now the readme indicates that this config file should work with avrdude version 6.3.  Version 6.3 doesn't have support for jtagice3_updi. That programmer type was added after the release. 
( jtagice3_updi was added in 2017 (http://svn.savannah.gnu.org/viewvc/avrdude/trunk/avrdude/ChangeLog-2017?view=log), avrdude 6.3 was released in 2016 (latest changelog: http://svn.savannah.gnu.org/viewvc/avrdude/tags/RELEASE_6_3/ChangeLog?view=log). )

Would you like me to push a config file that allows avrdude 6.3 to be used? It would be nice if the user can choose the config file depending on what version of avrdude they have installed.

ElTangas, 2020-06-19:
> Would you like me to push a config file that allows avrdude 6.3 to be used? It would be nice if the user could choose the config file depending on what version of avrdude they have installed.

Yeah, sure, but all these versions are called "6.3" even though they are different builds with different features. How to explain the nuances to users?

maartenvanschagen, 2020-06-20:
I agree, they made it very confusing. Maybe the terminology 'stable' (2016) and 'latest' (2020) makes more sense. Or 'release' and 'trunk' / 'dev' / 'development' / 'svn'?
It looks like most Linux distros ship "release" version 6.3 or the "dev" version released in 2017. "release" version 6.3 (from 2016) is also the latest release available in the download area (http://download.savannah.gnu.org/releases/avrdude/).

I'll make sure my edited config file is correct, it should be done in 3 days tops. I'll maybe experiment with a config file that makes use of the -C + option, perhaps I can get something to work on both versions.

closed 2020-07-03 by maartenvanschagen