ElTangas/jtag2updi #37: -C option

maartenvanschagen, 2020-06-17:
Hey ElTangas,

I have a request that would make the project more accessible to AVR newbies.

Avrdude allows you to specify a different configuration file path with the -C command. It would be very helpful to add this to the readme, maybe even with an example.

If you want to you could even add a config file to the repo with only the parts you added. This config file can be combined with the systemwide config by adding a + sign in front of the specified path.
Deleting everything that came before the following comment worked for me (and making the changes that I specified in my other issue):
`#------------------------------------------------------------`
`# AVR8X family common values`
`#------------------------------------------------------------`

ElTangas, 2020-06-19:
Yeah that readme really needs an update for many reasons...
I do assume a certain familiarity with avrdude.

maartenvanschagen, 2020-06-19:
> Yeah that readme really needs an update for many reasons...
> I do assume a certain familiarity with avrdude.

That sounds fair.

And by the way. Thank you very much for providing this project free and open source. It helps a lot to have such a good and affordable updi programming solution available.

closed 2020-06-19 by maartenvanschagen