ElTangas/jtag2updi #38: Where exactly does the UPDI transmission start and end in the code?

mikrocoder, 2020-06-19:
Hello,

first of all, thanks for maintaining the code. However, it is difficult to keep track of the flow of the actual code. :-) So I have to ask where in the code does the UPDI transmission start and end exactly?

I moved the switching on and off of the programming led.
I moved "on" to the beginning of enter_progmode() and "off" to the end of go(). But the beginning of the transmission is still missing.

Where exactly does the UPDI transmission start and end?

https://www.directupload.net/file/d/5855/hz7c22vr_png.htm

ElTangas, 2020-06-19:
There is a variable meant to keep track of that, JTAG2::ConnectedTo bit 0 is a flag that is set when UPDI is transmitting, cleared when not.
Anyway, the first time UPDI is used is in the JTAG2::set_device_descriptor() function, and the last is JTAG2::go() but I think the best way to keep track is the flag I mentioned.

mikrocoder, 2020-06-19:
Thank you very much.