ElTangas/jtag2updi #40: avrdude.conf is broken

mikrocoder, 2020-06-25:
Hello,

your current avrdude.conf is broken. The previous one without ATmega-DA support works.

For example. 
avrdude -c jtag2updi -P com9 -p m4808 -v

_avrdude: Version 6.3, compiled on Feb 17 2016 at 09:25:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\avrToolchain\avrdude\avrdude.conf"
avrdude: error at C:\avrToolchain\avrdude\avrdude.conf:1115: programmer type jtagice3_updi not found
avrdude: error reading system wide configuration file "C:\avrToolchain\avrdude\avrdude.conf"_

Something's not right.   :-)

guillermofbriceno, 2020-06-26:
Try using this one from their recent release candidate, they're working on it: [https://github.com/ElTangas/jtag2updi/pull/39/commits/3bdfecca297f874ca0f11871bf3dc33bcb5bdfb1](
https://github.com/ElTangas/jtag2updi/pull/39/commits/3bdfecca297f874ca0f11871bf3dc33bcb5bdfb1)

mikrocoder, 2020-06-27:
Thanks for this information.

ElTangas, 2020-07-02:
The problem is that avrdude.conf was updated from the one that comes with the arduino IDE.
The version of avrdude that comes with Arduino has certain recent changes patched into it, so it can handle this avrdude.conf file, but other builds can't. Sorry I didn't notice this :-(

ElTangas, 2020-07-02:
I have downloaded the base version of avrdude 6.3 from here: http://download.savannah.gnu.org/releases/avrdude/
I will use this to clean up avrdude.conf
Meanwhile, user @maartenvanschagen supplied a compatible version named "avrdude6.3.conf" that I will leave until I finish the cleanup.

mikrocoder, 2020-07-03:
Hello,

the problem has nothing to do with the Arduino IDE for now. It does not know my ATmega4808 anyway. Unless I add this one. I sometimes use avrdude (from the same source) standalone with your extended avrdude.conf, that's why I noticed that. I'm using your old version until this is fixed. What's messed up, if I may ask? To understand the problem. Thanks for taking care of it. High praise.

ElTangas, 2020-07-04:
The problem is that there are 2 major builds of avrdude 6.3:
The base build from here http://download.savannah.gnu.org/releases/avrdude/

and the patched build from the Arduino IDE
windows: http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-i686-w64-mingw32.zip
linux: http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-i686-pc-linux-gnu.tar.bz2
mac: http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-x86_64-apple-darwin12.tar.bz2
(you can extract these addresses from https://github.com/arduino/Arduino/blob/master/hardware/package_index_bundled.json)

If you use the base build, my current avrdude.conf will not work, if you use the Arduino build it will.
This is because the Arduino build has more recent features patched in, even though they are both "version 6.3".
I will remove dependency on these features to restore compatibility with base avrdude.

ElTangas, 2020-07-04:
I pushed a new version of avrdude.conf, that should solve the problem, please test.

mikrocoder, 2020-07-05:
This is very interesting information. I didn't know until today that avrdude.conf is adapted to the Arduino IDE.

I have tested your new avrdude.conf with my avrdude 6.3 standalone. ATmega4808. Works. Thanks a lot.

I will leave the original Arduino version alone for now. It works. The IDE Addon Package "MegaCoreX" from MCUdude also works.

Only what I don't know is, what or for whom is the Arduino version of your avrdude.conf intended? Or is it a template for Arduino.cc or people like MCUdude or Spencekonde etc.? 

By the way, I noticed something else when comparing. In your two new versions, part parent is ".avr8x_tiny" instead of ".avr8x_mega" for ATmega808 to ATmega1609.

mikrocoder, 2020-07-05:
This is very interesting information. I didn't know until today that avrdude.conf is adapted to the Arduino IDE.

I have tested your new avrdude.conf with my avrdude 6.3 standalone. ATmega4808. Works. Thanks a lot.

I will leave the original Arduino version alone for now. It works. The IDE Addon Package "MegaCoreX" from MCUdude also works.

Only what I don't know is, what or for whom is the Arduino version of your avrdude.conf intended? Or is it a template for Arduino.cc or people like MCUdude or Spencekonde etc.?

When comparing, I noticed that in the two new versions the entry for part parent for ATmega808 to ATmega1609 is ".avr8x_tiny". Shouldn't that be ".avr8x_mega"?

ElTangas, 2020-07-06:
> When comparing, I noticed that in the two new versions the entry for part parent for ATmega808 to ATmega1609 is ".avr8x_tiny". Shouldn't that be ".avr8x_mega"?

No. The entries ".avr8x_tiny" and ".avr8x_mega" differ only in the size of the "user row" memory area, which is the size of one EEPROM page.
So basically, every chip that has EEPROM page size of 32 bytes is ".avr8x_tiny" and the ones that have EEPROM page size of 64 bytes are ".avr8x_mega" regardless of their actual model names.

> Or is it a template for Arduino.cc or people like MCUdude or Spencekonde etc.?

That's the main reason, yes. Another one is that I want to raise peoples consciousness to the fact there are different versions of avrdude 6.3 that have different features. As you said,

> This is very interesting information. I didn't know until today that avrdude.conf is adapted to the Arduino IDE.

Like you, most people are unaware of these details, but I think they should be, because they are indeed interesting and useful to know.
I'll have to write about it in the documentation one day, when I finally decide to update it.