ElTangas/jtag2updi #42: Supported chips to use as programmer?

sideswiper22, 2020-07-24:
Hi! Aside from the atmega328p and 2560, what other similar avr chips are supported as a programmer? Will this work with attinies as well?

ElTangas, 2020-07-24:
Sorry, the documentation really needs an update...
Anyway, you can find a list of supported parts on this file:
https://github.com/ElTangas/jtag2updi/blob/master/source/parts.h

sideswiper22, 2020-07-25:
Thank you! There seems to be no list of what pin is used for the programming line. Where can I find the pin assignment and how can they be reassigned?

ElTangas, 2020-07-25:
The configuration is mostly done in this file:
https://github.com/ElTangas/jtag2updi/blob/master/source/sys.h

sideswiper22, 2020-08-09:
Any chance for the old ATtiny parts to be supported?

ElTangas, 2020-08-09:
> Any chance for the old ATtiny parts to be supported?

It's not likely, because those parts usually have small amounts of RAM, and jtag2updi requires at least 300 bytes.
This means only chips with 512 bytes of RAM or more can run it.

token47, 2023-02-26:
I have a lot of small boards with ATtiny85 and a USB port on them (with the size of a usb flash drive). It would be very nice to be able to run jtag2updi on them (the attiny85 does have 512 bytes of ram and 8k of flash). I think the main issue is that this board (and many other boards with the tiny series) do not have any USB-to-Serial chip, and would require the integration with V-USB, which although probably possible, is not necessarily easy. The ease with the arduino boards is that they come with the serial part ready to use.