ElTangas/jtag2updi #43: Can't unlock a locked ATtiny404 with chip erase - "avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ ILLEGAL_MCU_STATE"

willlewin, 2020-08-17:
Hi ElTangas,

Firstly, thanks for providing the library, it's great!

I'm just having this one issue when trying to unlock some ATtiny404s running at 1MHz clock using an Arduino Nano for the jtag2updi link. I've tried on a few chips now so I'm reasonably confident it's not hardware issues with the chips and is probably something I've overlooked. I compiled avrdude.conf for these at 1MHz and have been using it to flash hex files and lock them successfully.

I'm locking them with:
`avrdude -c jtag2updi -P com7 -p attiny404 -u -U lock:w:0xC4:m`
And trying to unlock with:
`avrdude -c jtag2updi -P com7 -p attiny404 -u -e`
Although I have tried a few other options and lock bit patterns.

Do you have any advice or thoughts on what is going on?

I've included the verbose output below:

```
"I:\Clients\066-UNITED BEV\066-03 HANDS FREE\Manufacturing\1D\01-software>"C:\Pro
gram Files (x86)\Arduino\hardware\tools\avr\bin\avrdude" -c jtag2updi -P com7 -p
 attiny404 -u -e -vvvv

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "I:\Clients\066-UNITED BEV\066-03 HAN
DS FREE\Manufacturing\1D\01-software\avrdude.conf"

         Using Port                    : com7
         Using Programmer              : jtag2updi
avrdude: jtagmkII_open_pdi()
avrdude: jtagmkII_getsync()
avrdude: jtagmkII_getsync(): Sending sign-on command:
avrdude: jtagmkII_send(): sending 1 bytes
avrdude: Send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] .
[f3] . [97]
avrdude: jtagmkII_recv():
avrdude: Recv: . [1b]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [1d]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [0e]
avrdude: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] .
[01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49
] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]
avrdude: Recv: < [3c]
avrdude: Recv: . [7f]

avrdude: jtagmkII_recv(): Got message seqno 0 (command_sequence == 0)
avrdude: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] .
[01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49
] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]

Raw message:
0x86 0x01 0x01 0x00 0x06 0x01 0x01 0x00 0x06 0x01 0x00 0x00 0x00 0x00 0x00 0x00
0x4a 0x54 0x41 0x47 0x49 0x43 0x45 0x20 0x6d 0x6b 0x49 0x49 0x00
Sign-on succeeded

JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
[... 534 lines cut ...]
avrdude: Send: . [1b] . [0e] . [00] . [01] . [00] . [00] . [00] . [0e] . [00] w
[77] . [b6]
avrdude: jtagmkII_recv():
avrdude: Recv: . [1b]
avrdude: Recv: . [0e]
avrdude: Recv: . [00]
avrdude: Recv: . [01]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [00]
avrdude: Recv: . [0e]
avrdude: Recv: . [80]
avrdude: Recv: . [7f]
avrdude: Recv: 2 [32]

avrdude: jtagmkII_recv(): Got message seqno 14 (command_sequence == 14)
avrdude: Recv: . [80]

Raw message:
0x80
OK

avrdude done.  Thank you.
```
`

ElTangas, 2020-08-19:
Yes, the -e option doesn't work correctly.
To unlock the chip, you need to enter terminal mode using -t option, and also -F to ignore errors.
Then in terminal mode enter the "erase" command, this should unlock the chip.

willlewin, 2020-08-21:
OK, it's working for me now. 

I had missed the build options required for terminal mode from the documentation when I tried this the first time. RTFM

Thanks so much for the support!

closed 2020-08-21 by willlewin

llinjupt, 2022-02-21:
IMHO，it seems like a bug of avrdude.  To erase without programming with '-e' should work, but avrdude always tries to enter progmode if no '-t'.  So here is a dealock case if the chip is locked, it 's impossibe to enter progmode.