ElTangas/jtag2updi #44: erase not working after lockbit set

mechatroniks-git, 2021-03-11:
Chip: Attiny 404

1. I programmed my device with the following `.......avrdude.conf -v -pattiny404 -cjtag2updi -PCOM1 -Ufuse2:w:0x02:m -Ufuse6:w:0x04:m -Ufuse8:w:0x00:m -Ulock:w:0x5C:m -Uflash:w:C:\Users.......`
2. I confirmed device is locked
3. I attempted to erase, but I can't seem to get it to erase

`C:\Users\m\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -C C:\Users\m\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.9/avrdude.conf -v  -p t404 -c jtag2updi -P COM1  -e

avrdude.exe: Version 6.3-20201216
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Users\m\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.9/avrdude.conf"

             Using Port                    : COM1
             Using Programmer              : jtag2updi
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
             AVR Part                      : ATtiny404
             Chip Erase delay              : 0 us
             PAGEL                         : P00
             BS2                           : P00
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 0
             StabDelay                     : 0
             CmdexeDelay                   : 0
             SyncLoops                     : 0
             ByteDelay                     : 0
             PollIndex                     : 0
             PollValue                     : 0x00
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
               prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
               fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
               fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
               data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
               usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
               flash          0     0     0    0 no       4096   64      0     0     0 0x00 0x00
               eeprom         0     0     0    0 no        128   32      0     0     0 0x00 0x00

             Programmer Type : JTAGMKII_PDI
             Description     : JTAGv2 to UPDI bridge
             M_MCU hardware version: 1
             M_MCU firmware version: 6.00
             S_MCU hardware version: 1
             S_MCU firmware version: 6.00
             Serial number:          00:00:00:00:00:00
             Vtarget         : 5.0 V

avrdude.exe: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude.exe: jtagmkII_reset(): timeout/error communicating with programmer (status -1)
avrdude.exe: initialization failed, rc=-1
             Double check connections and try again, or use -F to override
             this check.

avrdude.exe: jtagmkII_close(): timeout/error communicating with programmer (status -1)
avrdude.exe: jtagmkII_close(): timeout/error communicating with programmer (status -1)

avrdude.exe done.  Thank you.`

DaleMitchell, 2021-03-12:
To erase a microcontroller that is locked, the `-F` option must be specified. For example:
```
`C:\Users\m\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -C C:\Users\m\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.9/avrdude.conf -v -p t404 -c jtag2updi -P COM1 -e -F
```

mechatroniks-git, 2021-03-12:
Thank you for the clarification, I tried that on the first board. Didn't work.

So I took a 2nd  board, erased it, programmed it with lockbit set, then tried to erase, but it's locked still. 

```
C:\Users\m\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -C C:\Users\m\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.9/avrdude.conf -p t404 -c jtag2updi -P COM12 -e -F -v

avrdude.exe: Version 6.3-20201216
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Users\m\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.9/avrdude.conf"

             Using Port                    : COM12
             Using Programmer              : jtag2updi
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
             AVR Part                      : ATtiny404
             Chip Erase delay              : 0 us
             PAGEL                         : P00
             BS2                           : P00
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 0
             StabDelay                     : 0
             CmdexeDelay                   : 0
             SyncLoops                     : 0
             ByteDelay                     : 0
             PollIndex                     : 0
             PollValue                     : 0x00
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
               prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
               fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
               fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
               data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
               usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
               flash          0     0     0    0 no       4096   64      0     0     0 0x00 0x00
               eeprom         0     0     0    0 no        128   32      0     0     0 0x00 0x00

             Programmer Type : JTAGMKII_PDI
             Description     : JTAGv2 to UPDI bridge
             M_MCU hardware version: 1
             M_MCU firmware version: 6.00
             S_MCU hardware version: 1
             S_MCU firmware version: 6.00
[... 145 lines cut ...]
               data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
               usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
               flash          0     0     0    0 no       4096   64      0     0     0 0x00 0x00
               eeprom         0     0     0    0 no        128   32      0     0     0 0x00 0x00

             Programmer Type : JTAGMKII_PDI
             Description     : JTAGv2 to UPDI bridge
             M_MCU hardware version: 1
             M_MCU firmware version: 6.00
             S_MCU hardware version: 1
             S_MCU firmware version: 6.00
             Serial number:          00:00:00:00:00:00
             Vtarget         : 5.0 V

avrdude.exe: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude.exe: jtagmkII_reset(): bad response to reset command: RSP_ILLEGAL_MCU_STATE
avrdude.exe: initialization failed, rc=-1
             Double check connections and try again, or use -F to override
             this check.

avrdude.exe: jtagmkII_close(): bad response to sign-off command: RSP_ILLEGAL_MCU_STATE

avrdude.exe done.  Thank you.

```

DaleMitchell, 2021-03-12:
Maybe try a different UPDI programmer? I use this one, https://github.com/MCUdude/microUPDI
This works for me:

```C:\Users\dalem\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\dalem\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.9/avrdude.conf -v -pattiny806 -cxplainedmini_updi -e -F```

ElTangas, 2021-03-18:
Try to erase from interactive mode, i.e. use -t -F then run the "erase" command.

mechatroniks-git, 2021-03-29:
Interactive mode works, thank you.

closed 2021-03-29 by mechatroniks-git

mechatroniks-git, 2021-03-29:
Maybe I leave this open?

With the lock fuse set, the device ID as read did not match expected value. I think the -F does not ignore this?