ElTangas/jtag2updi #45: Error programming ATTiny1614: bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE

kais58, 2021-03-15:
Recieved the above error and others trying to program several MCUs, all same error, confirmed everyting I can is correct physically and couldn't find anything, full error output below. Using an Arduino Pro mini and standard setup withe resistor on UDPI line.

/home/kais58/.platformio/packages/tool-avrdude-megaavr/avrdude -C /home/kais58/.platformio/packages/tool-avrdude-megaavr/avrdude.conf -c jtag2updi -P /dev/ttyUSB0 -p attiny1614 -v -v -v -v

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/kais58/.platformio/packages/tool-avrdude-megaavr/avrdude.conf"
         User configuration file is "/home/kais58/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : jtag2updi
avrdude: jtagmkII_open_pdi()
avrdude: jtagmkII_getsync()
avrdude: jtagmkII_getsync(): Sending sign-on command: 
avrdude: jtagmkII_send(): sending 1 bytes
avrdude: Send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97] 
avrdude: jtagmkII_recv():
avrdude: Recv: . [1b] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [1d] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [0e] 
avrdude: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00] 
avrdude: Recv: < [3c] 
avrdude: Recv: . [7f] 

avrdude: jtagmkII_recv(): Got message seqno 0 (command_sequence == 0)
avrdude: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00] 

Raw message:
0x86 0x01 0x01 0x00 0x06 0x01 0x01 0x00 0x06 0x01 0x00 0x00 0x00 0x00 0x00 0x00
0x4a 0x54 0x41 0x47 0x49 0x43 0x45 0x20 0x6d 0x6b 0x49 0x49 0x00 
Sign-on succeeded

JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
avrdude: jtagmkII_getsync(): Using a 298-byte device descriptor
avrdude: jtagmkII_setparm()
avrdude: jtagmkII_setparm(): Sending set parameter command (parm 0x03, 1 bytes): 
avrdude: jtagmkII_send(): sending 3 bytes
avrdude: Send: . [1b] . [01] . [00] . [03] . [00] . [00] . [00] . [0e] . [02] . [03] . [06] $ [24] 1 [31] 
avrdude: jtagmkII_recv():
avrdude: Recv: . [1b] 
avrdude: Recv: . [01] 
avrdude: Recv: . [00] 
avrdude: Recv: . [01] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [0e] 
avrdude: Recv: . [80] 
avrdude: Recv: . [cd] 
avrdude: Recv: . [83] 

[... 394 lines cut ...]

avrdude: jtagmkII_close(): Sending sign-off command: 
avrdude: jtagmkII_send(): sending 1 bytes
avrdude: Send: . [1b] . [0e] . [00] . [01] . [00] . [00] . [00] . [0e] . [00] w [77] . [b6] 
avrdude: jtagmkII_recv():
avrdude: Recv: . [1b] 
avrdude: Recv: . [0e] 
avrdude: Recv: . [00] 
avrdude: Recv: . [01] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [0e] 
avrdude: Recv: . [80] 
avrdude: Recv: . [7f] 
avrdude: Recv: 2 [32] 

avrdude: jtagmkII_recv(): Got message seqno 14 (command_sequence == 14)
avrdude: Recv: . [80] 

Raw message:
0x80 
OK

avrdude done.  Thank you.

ElTangas, 2021-03-18:
Are you sure the chips aren't locked?

Frogieder, 2021-11-26:
I face the same issue. Different chip model yet similar error. My chips were brand new, no one used before, so I doubt they'd be locked. Also, I tried erasing the chip to unlock it, but that generated the same error

kosilin, 2022-03-05:
If locked these chips can be erased or reprogrammed by high-voltage programmer only. The PA0 pin is not in UPDI mode on locked chips until high-voltage pulse has applied to it.