ElTangas/jtag2updi #46: ATtiny1607 at 8MHz with Arduino IDE

Alailek, 2021-03-21:
Hi. I'm using the Arduino core megaTinyCore from SpenceKonde to compile the code with the Arduino IDE for an ATtiny1607. At 20MHz and 16MHz, everything's OK but below that, it compiles but it's not working. F_CPU is set correctly. At 8 or 10MHz, I need to double the baud rate of the host to be able to communicate with the program but it blocks when it tries to communicate with the target. 

Any idea of the problem?

Thanks

Alailek, 2021-03-21:
I just found out that the prescaler of the clock is disabled by the code in sys.cpp for the ATtiny 0 and 1 series

    #else //0-series or 1-series
      _PROTECTED_WRITE(CLKCTRL.MCLKCTRLB, 0);

By setting the MCLKCTRLB register to 1, I can use the clock to 8 or 10MHz setting in the Arduino IDE.

ElTangas, 2021-03-22:
Indeed, max clock is hard-coded for these chips in the source code. Maybe I should change it to get the right value according to F_CPU.

Alailek, 2021-03-22:
I want to reduce the clock speed because, according to the specification, the minimum voltage to run at maximum clock speed is 4.5 volts. At 8/10MHz, the minimum voltage is 2.7 volts (I want to use 3.3 volts). It would be nice to get the prescaler set automatically but since I know how to override it stays a "nice to have".

Thanks for your work!