ElTangas/jtag2updi #47: Why PD6?

jirkaptr, 2021-04-20:
When using a programmer made from ARDUINO UNO, NANO and possibly LEONARDO it would be convenient to use the connector originally designed for ICSP. For this case it would be necessary to use one of the terminals MISO, MOSI, SCK instead of PD6. What conditions would have to be met?

ElTangas, 2021-04-28:
You can change the default values to others, if you prefer.

homer1013, 2023-10-23:
I also had this question and started digging in to the code.

I have changed from Pin 6 (PD6) to Pin 11 (PB3) in order to utilize the ICSP pin header.

If you'd like to do so also, go to the "sys.h" file and find the arduino family for your project, ie. Arduino Uno R3 (ATmega328 P)

For my example I am using an Uno R3, see below for what's changed.

#elif defined (__AVR_ATmega_Mini__) || defined(ARDUINO_AVR_LARDU_328E)
// For ATmega8/88/168/328 (P, PB) parts

# ifndef HVLED_PORT
  #   define HVLED_PORT B
  # endif

  # ifndef HVLED_PIN
  #   define HVLED_PIN 0
  # endif

  #	ifndef HOST_USART
  #		define HOST_USART 0
  #	endif

  #	ifndef UPDI_PORT
  #		define UPDI_PORT B  // Pin 11 (PB3) ADDED
  //#		define UPDI_PORT D  // Pin 6 (PD6) ORIGINAL 
  #	endif

  #	ifndef UPDI_PIN
  #		define UPDI_PIN 3   // Pin 11 (PB3) ADDED
  //#		define UPDI_PIN 6   // Pin 6 (PD6) ORIGINAL
  #	endif

I hope this helps out anyone stuck right now
-Homer