ElTangas/jtag2updi #48: Arduino Nano Every as programmer

bdoubleu86, 2021-05-12:
Thanks for the great repository! I  do not have an issue, but a question. I am wondering whether it will also work with an Arduino Nano Every (ATMEGA4809) instead of an Arduino Nano as a programmer.

ElTangas, 2021-05-17:
Well, the mega4809 is supported, so it should work on the nano Every. I can't be absolutely sure since I never tested it on an Every.
The default pin for the UPDI interface is PB0 which corresponds to D9 on the nano every.
You can change those settings on "sys.h" if you want to. In fact I recommend that you change the LED pin to E2 which is where the nano every user LED is located.

I.e. find this in "sys.h"

> #elif defined (__AVR_ATmega_Zero__ ) || defined( __AVR_DA__)
// 4808, 4809. and the rest of the megaAVR 0-series
// Same as above, pretty much
// big difference here is your specify the name of the peripheral instead of the number, and the target baud rate, because we grab the OSCCAL value per datasheet./
//  #define USE_USARTDEBUG
//  #define DEBUG_USART USART1
//  #define DEBUG_BAUDRATE 2000000UL

and add this right after:

>#define LED_PORT E
#define LED_PIN 2

rocketscream, 2022-07-01:
I think the comment for:

```
#elif defined (__AVR_ATmega_Zero__)
// 4808, 4809. and the rest of the megaAVR 0-series
```
is not that accurate and as it is only valid for the 48-pin variant (eg: 4809) as on ATMega4808, it lacks PORTB.
So, it would be better if a separate definition is added like:
```
#elif defined (__AVR_ATmegax08__)
// 808, 4808 and the rest of the megaAVR 0-series with 32-pin and 28-pin package

  # ifndef UPDI_PORT
  #   define UPDI_PORT C
  # endif

  # ifndef UPDI_PIN
  #   define UPDI_PIN 0
  # endif

  # ifndef LED_PORT
  #   define LED_PORT A
  # endif

  # ifndef LED_PIN
  #   define LED_PIN 7
  # endif

  # ifndef HOST_USART
  #   define HOST_USART USART0
  # endif

  # ifndef HOST_TX_PORT
  #   define HOST_TX_PORT A
  # endif

  # ifndef HOST_TX_PIN
  #   define HOST_TX_PIN 0
  # endif

  # ifndef HOST_RX_PIN
  #   define HOST_RX_PIN 1
  # endif
```
Is there any specific requirements for the UPDI port/pin? I'm basically using the MegaCoreX as a base here where the LED pin is normally on PA7.

21846006, 2022-09-29:
> Thanks for the great repository! I do not have an issue, but a question. I am wondering whether it will also work with an Arduino Nano Every (ATMEGA4809) instead of an Arduino Nano as a programmer.

Hi Have you gotton the Nanp Every to work as a programmer?

colinluthier, 2023-03-21:
A changed the LED pin based on the instructions above and am trying to use an Nano Every to program a ATMega4809. When I upload jtag2updi to the Nano Every I am getting this error

`avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description`

colinluthier, 2023-03-21:
I am finding that this is a spurious error that can be ignored. When I ignore it I get this error uploading:

```avrdude warning: attempt 10 of 10: sign-on command: status -1
avrdude error: timeout/error communicating with programmer (status -1)
avrdude error: unable to open programmer jtag2updi on port COM5
Failed programming: uploading error: exit status 1

COM5 is the correct port. Is this indicating an error between the Nano and the 4809 chip?