ElTangas/jtag2updi #53: Unable to erase ATtiny402

Frogieder, 2021-11-26:
``I tried to communicate with the ATtiny402 over UPDI, yet I couldn't. My chips are brand new, I bought them from a very reliable seller. The error message I got after I managed to get the programmer working seemed similar to the message I am supposed to get with a locked chip. I tried to unlock it by erasing it using `-e` flag or by using `-F -t` flags, yet none of them worked. I was stuck with the same error message. I am sure that my connections are right, I added a 10uF capacitor on my Arduino Uno to prevent it from resetting (which indeed works), I have copied the `.conf` file, and all the code I'm using is the most recent one I can get to.  I am running everything on a Linux machine. There doesn't seem to be anything about this particular problem in other issues, not open nor closed, as far as I can tell. I'd really appreciate if you could look into this.
The outputs (both verbose and short) of my commands look like this:

$ avrdude -c jtag2updi -P /dev/ttyACM0 -p t402 -F -t

        avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
        avrdude: AVR device initialized and ready to accept instructions

        Reading |                                                    | 0% 0.00savrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE
        avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE
        avrdude: jtagmkII_read_byte(): bad response to read memory command: RSP_ILLEGAL_MCU_STATE
        avr_read(): error reading address 0x0000
            read operation not supported for memory "signature"
        avrdude: error reading signature data for part "ATtiny402", rc=-2
        avrdude: error reading signature data, rc=-2
        avrdude: jtagmkII_program_disable(): bad response to leave progmode command: RSP_ILLEGAL_MCU_STATE

        avrdude done.  Thank you.`

$ avrdude -c jtag2updi -P /dev/ttyACM0 -p t402 -F -t -v -v -v -v

    avrdude: Version 6.3-20171130
            Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
            Copyright (c) 2007-2014 Joerg Wunsch

            System wide configuration file is "/etc/avrdude.conf"
            User configuration file is "/home/martin/.avrduderc"
            User configuration file does not exist or is not a regular file, skipping

            Using Port                    : /dev/ttyACM0
            Using Programmer              : jtag2updi
    avrdude: jtagmkII_open_pdi()
    avrdude: jtagmkII_getsync()
    avrdude: jtagmkII_getsync(): Sending sign-on command: 
    avrdude: jtagmkII_send(): sending 1 bytes
    avrdude: Send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97] 
    avrdude: jtagmkII_recv():
    avrdude: Recv: . [1b] 
    avrdude: Recv: . [00] 
    avrdude: Recv: . [00] 
    avrdude: Recv: . [1d] 
    avrdude: Recv: . [00] 
    avrdude: Recv: . [00] 
    avrdude: Recv: . [00] 
    avrdude: Recv: . [0e] 
    avrdude: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00] 
    avrdude: Recv: < [3c] 
    avrdude: Recv: . [7f] 

    avrdude: jtagmkII_recv(): Got message seqno 0 (command_sequence == 0)
    avrdude: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00] 

    Raw message:
    0x86 0x01 0x01 0x00 0x06 0x01 0x01 0x00 0x06 0x01 0x00 0x00 0x00 0x00 0x00 0x00
    0x4a 0x54 0x41 0x47 0x49 0x43 0x45 0x20 0x6d 0x6b 0x49 0x49 0x00 
    Sign-on succeeded

    JTAG ICE mkII sign-on message:
    Communications protocol version: 1
    M_MCU:
    boot-loader FW version:        1
    firmware version:              6.00
    hardware version:              1
    S_MCU:
    boot-loader FW version:        1
    firmware version:              6.00
    hardware version:              1
    Serial number:                   00:00:00:00:00:00
    Device ID:                       JTAGICE mkII
    avrdude: jtagmkII_getsync(): Using a 298-byte device descriptor
[... 412 lines cut ...]
    avrdude: jtagmkII_close(): Sending sign-off command: 
    avrdude: jtagmkII_send(): sending 1 bytes
    avrdude: Send: . [1b] . [0e] . [00] . [01] . [00] . [00] . [00] . [0e] . [00] w [77] . [b6] 
    avrdude: jtagmkII_recv():
    avrdude: Recv: . [1b] 
    avrdude: Recv: . [0e] 
    avrdude: Recv: . [00] 
    avrdude: Recv: . [01] 
    avrdude: Recv: . [00] 
    avrdude: Recv: . [00] 
    avrdude: Recv: . [00] 
    avrdude: Recv: . [0e] 
    avrdude: Recv: . [80] 
    avrdude: Recv: . [7f] 
    avrdude: Recv: 2 [32] 

    avrdude: jtagmkII_recv(): Got message seqno 14 (command_sequence == 14)
    avrdude: Recv: . [80] 

    Raw message:
    0x80 
    OK

    avrdude done.  Thank you.
    `

voneiden, 2021-11-26:
#45 appears to have same symptom?

Frogieder, 2021-11-26:
Oh, my bad, this really looks like duplicate of #45

closed 2021-11-26 by Frogieder

voneiden, 2021-11-26:
Would #43 help? Worth trying?

Frogieder, 2021-11-26:
Unfortunately not, as I said, I tried both options `-e` and `-t -F` yet both of them gave me the same error

Frogieder, 2021-11-26:
Alright, a big update: I managed to get into the terminal mode. I haven't uncommented a line in `sys.h` that I was supposed to. The same problem as in the issue you linked - I just didn't read anything. There is another error when I type in `erase` but there may be something I missed, so I'm going to do some research on my own again before I report a problem