ElTangas/jtag2updi #54: Suppressing avrdude harmless warning on "Cannot locate “flash” and “boot” memories in description"

d3lta-v, 2021-12-26:
Hi,

Thank you very much for writing this program as it has allowed both official Arduinos (e.g. Nano Every) and other AVRs to be able to take advantage of UPDI in a cost effective manner.

I am currently developing an educational microcontroller board (ATmega4809 based) that utilises `jtag2updi` for the programming interface and noticed that a spurious warning `avrdude: jtagmkII_initialize(): Cannot locate “flash” and “boot” memories in description` would come up every time code is being uploaded. 

I have also noticed this in other microcontrollers that used `jtag2updi` in an article [here](http://fabacademy.org/2021/labs/ingegno/students/cristiano-lemes/assignments/week11/).  Although this error is harmless, it impacts the end-user's (typically students) experience as they have to learn about the error and that it is not a fault of the device.

Would it be possible to suppress or eliminate this error in any particular configuration (e.g. through `avrdude.conf`)? I've did some digging and it appears that the error most likely originates from this section in the [avrdude configuration file](https://github.com/ElTangas/jtag2updi/blob/c76c70cb73e51abb6abab6c286edb68e37093582/avrdude.conf.arduino#L16189) which did not define the `memory "boot"` section, which (to my loose understanding) is not needed in the ATmega4809, but needed for older XMEGA series chips which the 4809 was based on (now called megaAVR 0-series). The relevant piece of code `avrdude` should be somewhere around [here](https://github.com/avrdudes/avrdude/blob/c60c5c4b83f737e366a2eb642be4aa74660e3bf7/src/jtagmkII.c#L1356)

E.g. an XMEGA chip with a defined boot section
[image]

E.g. ATmega4809 with a user-definable boot section
[image]

ElTangas, 2022-01-01:
Yes, I'm aware of this problem. Have you tested if a change to avrdude.conf fixes the warning?

d3lta-v, 2022-01-05:
Hi there,

I have tried creating a “dummy” boot section in the avrdude.conf for that particular microcontroller, but it will only result in an RSP_ILLEGAL_PARAMETER error being thrown during upload, presumably because it’s trying to read something from the MCU that doesn’t exist. 

From the looks of it I believe this issue will have to be resolved upstream eventually [in an avrdude pull request](https://github.com/avrdudes/avrdude/pull/789/commits/f96b98e9dfb99e422eadef71c97b44d60dceeb35) itself since the error stems from there. Since the issue is technically not affiliated with this repo, I think it’s safe to close it :)

closed 2022-01-05 by d3lta-v