ElTangas/jtag2updi #58: Which parts support NVM version 2?

llinjupt, 2022-02-24:
Hi All,

I wonder which parts support NVM version 2?   I have checked the code and found the NVM version depends on flash pagesize.
`nvm_version = (flash_pagesize >= 256) ? 2 : 1;`

But after glancing over those datasheets, even the largest size-mounted flash parts like ATtiny3224/3226/3227 only implemented with flash pagesize 128Bytes. So anyone knows which parts support NVM verson 2/ flash page size >= 256 bytes. Thanks in advance.

ElTangas, 2022-02-24:
These parts are the AVR-DA and AVR-DB.
You can check the end of the avrdude.conf file for a full list.

llinjupt, 2022-02-25:
> These parts are the AVR-DA and AVR-DB. You can check the end of the avrdude.conf file for a full list.
> 
> For example: https://www.microchip.com/en-us/product/AVR128DA48

Thanks for YOUR quick response, and that's really helpfull. BTW, the parts of AVR series are really making people confused. :<)

ElTangas, 2022-03-01:
Yeah, after Microchip bought Atmel, they came up with this new naming scheme.