ElTangas/jtag2updi #59: Unstable trailing edge of stopbit pulse while using software UPDI I/O

kosilin, 2022-03-05:
Good day!

Yesterday I've spend some time trying to make working UPDI HV mini programmer that I'd made using slightly modified schematics. I investigated that updi_io_soft.cpp contains an error (at my sight) that leads to either unstable or completely failed programmer operating.

The problem is that at stop bit transmission UPDI pin is switched to input mode (line 225, updi_io_soft.cpp) just after it has set to "1" (line 214, updi_io_soft.cpp). The trailing edge of the stop bit might not be detected by target device due to very short positive pulse that is damping completely as a fact because of parasitic capacitances. I think it's an issue.

I suggest to add some delay after stop bit transmission. I'm not sure but it seems to be reasonable for me to append the same time as "0" pulse has (txdelay).

At least in my environment the modification has led to stable operating of my UPDI programmer.

This patch is included.
[0001-Fixes-unstable-trailing-edge-of-stopbit-pulse-while-using-software-UPDI-IO.txt](https://github.com/ElTangas/jtag2updi/files/8190190/0001-Fixes-unstable-trailing-edge-of-stopbit-pulse-while-using-software-UPDI-IO.txt)

Best regards,
Alexey

ElTangas, 2022-03-12:
Thank you. Yes actually I did that on purpose at the time for some reason it seemed like a god idea :P
I think I will add your modification after some tests.

ElTangas, 2022-03-18:
Just an update, tests are progressing well, I've been able to reproduce your problem by using a 1k resistor and 500pF capacitance in the UPDI line.
A delay of 1/2 bit time seems to be enough, I will run some more tests over the next days.

kosilin, 2022-03-19:
Thanks! I'm glad that my investigation could help somebody.
But... 500pF seems to be quite large capacitance. What MCU are you using? And what clock frequency?
As for me, my programmer is basing on ATTiny1616 at 20MHz clock.

ElTangas, 2022-03-20:
> But... 500pF seems to be quite large capacitance. What MCU are you using? And what clock frequency?

I increased the capacitance until the problem appeared. It's an artificial experiment.
As you say, the problems start at about 200pF but not always in my case. So I used 500pF to make sure it always fails.

But with the fix it can work even with 1nF :) so thanks.

ElTangas, 2022-03-21:
Ok, I've updated the code. It's not exactly the same delay you used, but it should work.
I would be grateful if you could test it.

kosilin, 2022-03-21:
Thanks! I'll test the patch in a few days.

kosilin, 2022-03-22:
Hello!

I've just tested the patch - everything is ok. My opinion... At one hand the delay can be much less in principle. But at the other hand the target will not begin transmission before time of two stop bits has expired. So I think the applied delay is quite reasonable.

As an illustration i've included the  picture from my oscilloscope:
[image]

I think the issue is solved. Thank you very much!

Best regards,
Alexey

ElTangas, 2022-03-22:
Cool. I'll close it then.

closed 2022-03-22 by ElTangas