ElTangas/jtag2updi #63: Can't unlock ATTiny1624

Strat00s, 2022-05-20:
Well, I have no idea where else to go. I bought a brand new ATTiny from Mouser and it seemed to come locked. So I tried to unlock it by erasing it (`-e` or `-e -F`), which just gave me the error here:
```
avrdude.exe: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude.exe: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE
avrdude.exe: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE
avrdude.exe: jtagmkII_read_byte(): bad response to read memory command: RSP_ILLEGAL_MCU_STATE
avr_read(): error reading address 0x0000
    read operation not supported for memory "signature"
avrdude.exe: error reading signature data for part "ATtiny1624", rc=-2
avrdude.exe: error reading signature data, rc=-2
avrdude.exe: jtagmkII_program_disable(): bad response to leave progmode command: RSP_ILLEGAL_MCU_STATE

avrdude.exe done.  Thank you.
```

I then tried to use the interactive terminal. So I uncommented the `#define DISABLE_HOST_TIMEOUT` and `#define DISABLE_TARGET_TIMEOUT` (tried it with and without the second define). I was then able to get into the terminal. But when I tried to erase the chip from here, I just got another error:

```
avrdude.exe: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude.exe: jtagmkII_reset(): bad response to reset command: RSP_ILLEGAL_MCU_STATE
avrdude.exe: initialization failed, rc=-1
avrdude.exe: AVR device initialized and ready to accept instructions
avrdude.exe: Device signature = 0x666c61
avrdude.exe: Expected signature for ATtiny1624 is 1E 94 2A
avrdude> erase
>>> erase
avrdude.exe: erasing chip
avrdude.exe: jtagmkII_chip_erase(): timeout/error communicating with programmer (status -1)
avrdude> quit
>>> quit
avrdude.exe: jtagmkII_close(): timeout/error communicating with programmer (status -1)
avrdude.exe: jtagmkII_close(): timeout/error communicating with programmer (status -1)

avrdude.exe done.  Thank you.
```
Looks to be the same error as here https://github.com/ElTangas/jtag2updi/issues/24, which makes me guess, that my ATTiny is broken.

So, is my ATTiny actually broken or is there anything I can try? Thanks in advance for any help

<br/>
<br/>

Here is the verbose log when using `-e` only:
```
avrdude.exe: Version 6.3-20190619
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Users\Stratos\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.11/avrdude.conf"

             Using Port                    : COM7
             Using Programmer              : jtag2updi
             Overriding Baud Rate          : 115200
avrdude.exe: jtagmkII_open_pdi()
avrdude.exe: jtagmkII_getsync()
avrdude.exe: jtagmkII_getsync(): Sending sign-on command:
avrdude.exe: jtagmkII_send(): sending 1 bytes
avrdude.exe: Send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97]
avrdude.exe: jtagmkII_recv():
avrdude.exe: Recv: . [1b]
avrdude.exe: Recv: . [00]
avrdude.exe: Recv: . [00]
avrdude.exe: Recv: . [1d]
avrdude.exe: Recv: . [00]
avrdude.exe: Recv: . [00]
avrdude.exe: Recv: . [00]
avrdude.exe: Recv: . [0e]
avrdude.exe: Recv: . [86] . [01] . [01] . [00] . [06] . [01] . [01] . [00] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]
[... 375 lines cut ...]

avrdude.exe: jtagmkII_close(): Sending sign-off command:
avrdude.exe: jtagmkII_send(): sending 1 bytes
avrdude.exe: Send: . [1b] . [0a] . [00] . [01] . [00] . [00] . [00] . [0e] . [00] . [a9] . [a0]
avrdude.exe: jtagmkII_recv():
avrdude.exe: Recv: . [1b]
avrdude.exe: Recv: . [0a]
avrdude.exe: Recv: . [00]
avrdude.exe: Recv: . [01]
avrdude.exe: Recv: . [00]
avrdude.exe: Recv: . [00]
avrdude.exe: Recv: . [00]
avrdude.exe: Recv: . [0e]
avrdude.exe: Recv: . [80]
avrdude.exe: Recv: . [a1]
avrdude.exe: Recv: $ [24]

avrdude.exe: jtagmkII_recv(): Got message seqno 10 (command_sequence == 10)

Raw message:
0x80
OK

avrdude.exe done.  Thank you.
```

ElTangas, 2022-05-25:
Well, I think I never actually tested jtag2updi with a Tiny AVR-2 series chip. The UPDI unit seems identical to series 0 and 1 so it should work but I need to be sure now that you report a problem.

Anyway, another possible problem is that the UPDI pin is disabled (in reset or I(O mode). It should come in UPDI mode from the factory, but in this time of chip shortages there are some "strange" units being sold...

In this case all I can do is redirect you to https://github.com/Dlloydev/jtag2updi/wiki/DIY-ATtiny-HV-UPDI-Programmer
At the end of the page is a schematic for a HV programmer, you can try building one and use Dlloydev special version of jtag2updi to try to unlock the chip.

Sorry if can't be more helpful right now, I might need to get some Tiny AVR-2 samples for testing to make sure there isn't a bug with jtag2updi.

Strat00s, 2022-05-25:
Update

Tried a new chip and everything worked just fine. It was unlocked and the upload works as it should. So I guess that the first one is either damaged or locked. So I am gonna try the HV programmer, but don't know when I'll have time.

ElTangas, 2022-05-26:
Ok, good to know tiny AVR-2 can be programmed or I would have to go bug hunting...