ElTangas/jtag2updi #66: erasing locked AVR32DA28 - impossible

flexphperia, 2022-08-20:
I am having trouble erasing flash in AVR32DA28. I'm using Arduino UNO as UPDI programmer. Is there any chance to make it work?

I have locked MCU with command:
`avrdude -c jtag2updi -p avr32da28 -P COM10 -U lock:w:0xa3:m`

Now erasing returns error (also i cannot enter into terminal mode):
```
d:\jtag2updi-master>avrdude -c jtag2updi -p avr32da28 -P COM10 -u -e -F
avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE
avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE
avrdude: jtagmkII_read_byte(): bad response to read memory command: RSP_ILLEGAL_MCU_STATE
avr_read(): error reading address 0x0000
    read operation not supported for memory "signature"
avrdude: error reading signature data for part "AVR32DA28", rc=-2
avrdude: error reading signature data, rc=-2
avrdude: jtagmkII_program_disable(): bad response to leave progmode command: RSP_ILLEGAL_MCU_STATE

avrdude done.  Thank you.
```

ElTangas, 2022-08-21:
Hello, due to the way "-F" and "-e" options operate in avrdude, jtag2updi can't currently unlock chips that way.
You need to enter interactive mode "-F -t" and then use the "erase" command.

Note: You must build jtag2updi with DISABLE_HOST_TIMEOUT defined for terminal mode to work.

mcuee, 2022-10-17:
> Hello, due to the way "-F" and "-e" options operate in avrdude, jtag2updi can't currently unlock chips that way. You need to enter interactive mode "-F -t" and then use the "erase" command.
> 
> Note: You must build jtag2updi with DISABLE_HOST_TIMEOUT defined for terminal mode to work.

Indeed this trick works. I was hoping no need to rebuild with `DISABLE_HOST_TIMEOUT` option but it did not work.

```
PS C:\work\avr\avrdude_test\avrdude_bin> echo "erase"  | .\avrdude_git -c jtag2updi -P COM4 -p m4808 -qq -F -t
avrdude_git.exe: jtagmkII_reset(): bad response to reset command: RSP_ILLEGAL_MCU_STATE
avrdude_git.exe: initialization failed, rc=-1
avrdude_git.exe: Yikes!  Invalid device signature.
avrdude_git.exe: Expected signature for ATmega4808 is 1E 96 50
avrdude> >>> erase
avrdude_git.exe: erasing chip
avrdude>

PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump lock" | .\avrdude_git -c jtag2updi -P COM4 -p m4808 -qq  -t
avrdude> >>> dump lock
0000  c5                                                |.               |

avrdude>

```