ElTangas/jtag2updi #68: Arduino uno progrramer error

muhammetfaik, 2022-09-22:
I want to make programmer arudiono uno but I have this error:
```
Arduino: 1.8.19 (Linux), Board: "ATtiny1614/1604/814/804/414/404 w/Optiboot, ATtiny414, 20 MHz internal, Enabled (default timer), 1.8V (5 MHz or less), Disabled/Disabled, TX:5 (PB2), RX:4 (PB3), UPDI, Optiboot for 8 sec. after power-on & reset, 8ms, Master or Slave (saves flash and RAM), Default (doesn't print floats, 1.4k flash use), On all pins, like usual"

In file included from /home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/updi_io_soft.cpp:10:0:
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.h:326:4: warning: #warning "Part not supported - if you didn't provide all the needed pin definitions, that's why it's not compiling" [-Wcpp]
   #warning "Part not supported - if you didn't provide all the needed pin definitions, that's why it's not compiling"
    ^~~~~~~
In file included from /home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.cpp:10:0:
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.h:326:4: warning: #warning "Part not supported - if you didn't provide all the needed pin definitions, that's why it's not compiling" [-Wcpp]
   #warning "Part not supported - if you didn't provide all the needed pin definitions, that's why it's not compiling"
    ^~~~~~~
In file included from /home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/JTAG2.h:12:0,
                 from /home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/JICE_io.h:13,
                 from /home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/JICE_io.cpp:10:
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.h:326:4: warning: #warning "Part not supported - if you didn't provide all the needed pin definitions, that's why it's not compiling" [-Wcpp]
   #warning "Part not supported - if you didn't provide all the needed pin definitions, that's why it's not compiling"
    ^~~~~~~
In file included from /home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.cpp:11:0:
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/dbg.h:232:4: warning: #warning "Part not supported - if you didn't provide all the needed pin definitions, that's why it's not compiling" [-Wcpp]
   #warning "Part not supported - if you didn't provide all the needed pin definitions, that's why it's not compiling"
    ^~~~~~~
In file included from /home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/JICE_io.cpp:12:0:
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/dbg.h:232:4: warning: #warning "Part not supported - if you didn't provide all the needed pin definitions, that's why it's not compiling" [-Wcpp]
   #warning "Part not supported - if you didn't provide all the needed pin definitions, that's why it's not compiling"
    ^~~~~~~
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/updi_io_soft.cpp: In function 'uint8_t UPDI_io::get()':
sys.h:48:26: error: 'VPORTUPDI_PORT' was not declared in this scope
   # define DDR(x) CONCAT(VPORT,x).DIR
                          ^
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.h:39:21: note: in definition of macro 'CONCAT'
 #define CONCAT(A,B) A##B                // concatenate
                     ^
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/updi_io_soft.cpp:54:9: note: in expansion of macro 'DDR'
         DDR(UPDI_PORT)  &= ~(1 << UPDI_PIN);
         ^~~
In file included from /home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.cpp:10:0:
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.cpp: In function 'void SYS::init()':
sys.h:48:26: error: 'VPORTLED_PORT' was not declared in this scope
   # define DDR(x) CONCAT(VPORT,x).DIR
                          ^
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.h:39:21: note: in definition of macro 'CONCAT'
 #define CONCAT(A,B) A##B                // concatenate
                     ^
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.cpp:42:3: note: in expansion of macro 'DDR'
   DDR(LED_PORT) |= (1 << LED_PIN);
   ^~~
In file included from /home/muhammetubuntu/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/avr/include/avr/io.h:99:0,
                 from /home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/JICE_io.cpp:9:
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/JICE_io.cpp: In function 'uint8_t JICE_io::put(char)':
JICE_io.cpp:55:25: error: 'HOST_USART' was not declared in this scope
   loop_until_bit_is_set(HOST_USART.STATUS, USART_DREIF_bp);
                         ^
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.h:48:26: note: suggested alternative: 'VPORTA_OUT'
   # define DDR(x) CONCAT(VPORT,x).DIR
                          ^
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.h:39:21: note: in definition of macro 'CONCAT'
 #define CONCAT(A,B) A##B                // concatenate
                     ^
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.cpp:42:3: note: in expansion of macro 'DDR'
   DDR(LED_PORT) |= (1 << LED_PIN);
   ^~~
sys.cpp:42:26: error: 'LED_PIN' was not declared in this scope
   DDR(LED_PORT) |= (1 << LED_PIN);
                          ^~~~~~~
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.h:48:26: note: suggested alternative: 'VPORTA_OUT'
   # define DDR(x) CONCAT(VPORT,x).DIR
                          ^
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/sys.h:39:21: note: in definition of macro 'CONCAT'
 #define CONCAT(A,B) A##B                // concatenate
[... 161 lines cut ...]
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/JICE_io.h:18:12: note: in definition of macro 'loop_until_bit_set_or_host_timeout'
   while(!((register&(1<<bitpos))||(SYS::checkTimeouts() & WAIT_FOR_HOST))); \
            ^~~~~~~~
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/JICE_io.cpp:106:38: note: suggested alternative: 'IO_START'
   loop_until_bit_set_or_host_timeout(HOST_USART.STATUS, USART_TXCIF_bp);
                                      ^
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/JICE_io.h:18:12: note: in definition of macro 'loop_until_bit_set_or_host_timeout'
   while(!((register&(1<<bitpos))||(SYS::checkTimeouts() & WAIT_FOR_HOST))); \
            ^~~~~~~~
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/JICE_io.cpp: In function 'void JICE_io::set_baud(JTAG2::baud_rate)':
JICE_io.cpp:114:3: error: 'HOST_USART' was not declared in this scope
   HOST_USART.BAUD = baud_tbl[rate - 1];
   ^~~~~~~~~~
/home/muhammetubuntu/Codes/jtag2updi/source/jtag2updi/JICE_io.cpp:114:3: note: suggested alternative: 'IO_START'
   HOST_USART.BAUD = baud_tbl[rate - 1];
   ^~~~~~~~~~
   IO_START
exit status 1
'VPORTUPDI_PORT' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```
Help please!

ElTangas, 2022-09-27:
You are using attiny414 right? I'll look into it.

ElTangas, 2022-09-27:
Wait, you said you want to make a uno programmer, why did you select a tiny414 board?
First, you have to put jtag2updi in the uno, selecting the uno board.

Then you can program the tiny414 with the uno.