ElTangas/jtag2updi #69: jtag2updi broken with avrdude-7.0

mhx, 2022-12-01:
Just a quick heads-up that `jtag2updi` stopped working for me after upgrading to avrdude-7.0.

The failure mode is as follows:

```
% avrdude -p t824 -c jtag2updi -P /dev/ttyUSB2 -C avrdude.conf

Error: programmer jtag2updi does not support target ATtiny824

avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: jtagmkII_close(): bad response to GO command: RSP_NO_TARGET_POWER
avrdude: jtagmkII_close(): bad response to sign-off command: RSP_NO_TARGET_POWER

avrdude done.  Thank you.
```

`avrdude.conf` is the latest version from the `jtag2updi` master branch.

Here's the output with `-v`:

```
% avrdude -p t824 -c jtag2updi -P /dev/ttyUSB2 -C avrdude.conf -v

avrdude: Version 7.0
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "avrdude.conf"
         User configuration file is "/home/mhx/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB2
         Using Programmer              : jtag2updi
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
         AVR Part                      : ATtiny824
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9    1      0     0     0 0x00 0x00
           fuse0                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           usersig                 0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash                   0     0     0    0 no       8192   64      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        128   32      0     0     0 0x00 0x00

         Programmer Type : JTAGMKII_PDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU HW version: 1
         M_MCU FW version: 6.00
         S_MCU HW version: 1
         S_MCU FW version: 6.00
         Serial number   : 00:00:00:00:00:00
         Vtarget         : 5.0 V

Error: programmer jtag2updi does not support target ATtiny824

avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: jtagmkII_close(): bad response to GO command: RSP_NO_TARGET_POWER
avrdude: jtagmkII_close(): bad response to sign-off command: RSP_NO_TARGET_POWER

avrdude done.  Thank you.
```

Downgrading to avrdude-6.3 makes things work again:

```
% avrdude -p t824 -c jtag2updi -P /dev/ttyUSB2 -C avrdude.conf   

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e9329 (probably t824)

avrdude done.  Thank you.
```

I'm happy to do more tests / provide more information if necessary.

ElTangas, 2023-01-17:
Yes I'm aware of this thanks. Lately I've just been short on time to correct it :(

mhx, 2023-04-04:
Thanks, that actually works! :)

closed 2023-04-04 by mhx