ElTangas/jtag2updi #72: Building fail for Arduino Leonardo/ProMicro (Atmega32u4)

AKTanara, 2023-08-30:
I knew I'd have issues while trying to upload my code using Pro Mico as UPDI programmer, but I didn't think I'd have issues programming the programmer itself!
While compiling in Arduino IDE 1.8 for any of the boards (Leonardo, Pro Micro, Fio v3, ...) all having Atmega32U4 chip, returns this ERROR:

```
Compiling 'jtag2updi' for 'Arduino Leonardo'
Error compiling project sources
 
 
Build failed for project 'jtag2updi'
sys.cpp:10: In file included from
io.h:99: In file included from
JICE_io.cpp:9: from
 
 
sys.cpp: In function void SYS::init()
JICE_io.cpp: In function uint8_t JICE_io::put(char)
 
 
sys.h: 55:27: error: 'PORTUPDI_PORT' was not declared in this scope
sys.h: 346:32: error: 'UCSRHOST_USARTA' was not declared in this scope
   # define PORT(x) CONCAT(PORT,x)
   #define HOST_UCSRA XCONCAT(UCSR, XCONCAT(HOST_USART, A))
sys.h:39: note  in definition of macro CONCAT
sys.h:40: note  in expansion of macro CONCAT
   #define CONCAT(A,B) A##B                \\ concatenate
   #define XCONCAT(A,B) CONCAT(A,B)        \\ expand and concatenate
sys.cpp:38: note  in expansion of macro PORT
   ^~~~~~
   PORT(UPDI_PORT) = 1<<UPDI_PIN
sys.h:346: note  in expansion of macro XCONCAT
   ^~~~
   #define HOST_UCSRA XCONCAT(UCSR, XCONCAT(HOST_USART, A))
   ^~~~~~~
 
JICE_io.cpp:59: note  in expansion of macro HOST_UCSRA
   loop_until_bit_is_set(HOST_UCSRA, UDRE)
 
updi_io_soft.cpp:10: In file included from
   ^~~~~~~~~~
sys.cpp: 38:26: error: 'UPDI_PIN' was not declared in this scope
sys.h:346: note  suggested alternative  HOST_UCSRA
   PORT(UPDI_PORT) = 1<<UPDI_PIN
   #define HOST_UCSRA XCONCAT(UCSR, XCONCAT(HOST_USART, A))
   ^~~~~~~~
sys.h:40: note  in expansion of macro CONCAT
sys.cpp:38: note  suggested alternative  USB_PID
   #define XCONCAT(A,B) CONCAT(A,B)        \\ expand and concatenate
   PORT(UPDI_PORT) = 1<<UPDI_PIN
 
   ^~~~~~
   ^~~~~~~~
sys.h:346: note  in expansion of macro XCONCAT
   USB_PID
updi_io_soft.cpp: In function uint8_t UPDI_io::get()
   #define HOST_UCSRA XCONCAT(UCSR, XCONCAT(HOST_USART, A))
   ^~~~~~~
 
JICE_io.cpp:59: note  in expansion of macro HOST_UCSRA
 
sys.cpp:10: In file included from
   loop_until_bit_is_set(HOST_UCSRA, UDRE)
sys.h: 56:26: error: 'DDRUPDI_PORT' was not declared in this scope
   ^~~~~~~~~~
   # define DDR(x) CONCAT(DDR,x)
sys.h:39: note  in definition of macro CONCAT
 
   #define CONCAT(A,B) A##B                \\ concatenate
sys.h: 56:26: error: 'DDRLED_PORT' was not declared in this scope
 
updi_io_soft.cpp:54: note  in expansion of macro DDR
[... 408 lines cut ...]
JICE_io.cpp:92: note  in expansion of macro TXEN
   HOST_UCSRB = (1<<RXEN)|(1<<TXEN)
   ^~~~
 
JICE_io.cpp:10: In file included from
 
JICE_io.cpp: In function void JICE_io::flush()
 
sys.h: 346:32: error: 'UCSRHOST_USARTA' was not declared in this scope
   #define HOST_UCSRA XCONCAT(UCSR, XCONCAT(HOST_USART, A))
JICE_io.h:18: note  in definition of macro loop_until_bit_set_or_host_timeout
   checkTimeouts() & WAIT_FOR_HOST)))
   ^~~~~~~~
sys.h:40: note  in expansion of macro CONCAT
   #define XCONCAT(A,B) CONCAT(A,B)        \\ expand and concatenate
   ^~~~~~
sys.h:346: note  in expansion of macro XCONCAT
   #define HOST_UCSRA XCONCAT(UCSR, XCONCAT(HOST_USART, A))
   ^~~~~~~
JICE_io.cpp:108: note  in expansion of macro HOST_UCSRA
   loop_until_bit_set_or_host_timeout(HOST_UCSRA, TXC)
   ^~~~~~~~~~
sys.h:346: note  suggested alternative  HOST_UCSRA
   #define HOST_UCSRA XCONCAT(UCSR, XCONCAT(HOST_USART, A))
```

Obliman, 2023-10-19:
I had this same error and was able to get around it by editing the sys.h file to reflect the expected values from [iom32u4.h](https://github.com/avrdudes/avr-libc/blob/main/include/avr/iom32u4.h). The problem appears to be that the board is not recognized correctly, so "__AVR_ATmega32__" is not defined and then "__AVR_ATmega_Mighty__" is not defined, which leads to many other things not being defined properly...

Many definitions simply needed a "1" added to the name. Ex:
```
    #define UDRE UDRE1 // XCONCAT(UDRE, HOST_USART)
    #define U2X U2X1 // XCONCAT(U2X, HOST_USART)
    #define TXEN TXEN1 // XCONCAT(TXEN, HOST_USART)
    #define RXEN RXEN1 // XCONCAT(RXEN, HOST_USART)
    #define RXC RXC1 // XCONCAT(RXC, HOST_USART)
    #define TXC TXC1 // XCONCAT(TXC, HOST_USART)

    #define HOST_UCSRA UCSR1A // XCONCAT(UCSR, XCONCAT(HOST_USART, A))
    #define HOST_UCSRB UCSR1B // XCONCAT(UCSR, XCONCAT(HOST_USART, B))
    #define HOST_UDR UDR1 // XCONCAT(UDR, HOST_USART)

    #define HOST_UBRR UBRR1 // XCONCAT(UBRR, XCONCAT(HOST_USART, L))
```

However, I am still having communication issues with my board so I can't confirm whether this actually worked beyond stopping the error from appearing.

I suspect the port/pin definitions aren't right either, but I haven't figured it out what they should be yet:
```
# ifndef HOST_USART
#   define HOST_USART 0
# endif

# ifndef UPDI_PORT
#   define UPDI_PORT C
# endif

# ifndef UPDI_PIN
#   define UPDI_PIN 7
# endif

# ifndef LED_PORT
#   define LED_PORT B
# endif

# ifndef LED_PIN
#   define LED_PIN 7
# endif
```

Hopefully this helps the author create a fix.

Tomassmits, 2024-02-05:
@nhansendev Have you had any success in getting it to work? I'm attempting the same (on a DFR0282 Beetle) without much luck.

Tomassmits, 2024-02-09:
> > @nhansendev Have you had any success in getting it to work? I'm attempting the same (on a DFR0282 Beetle) without much luck.
> 
> Unfortunately, no. I decided it wasn't worth the hassle and switched to the typical Arduino Nano as my programmer instead.

Thanks. Same here; it's not worth the time, and the ATmega328 variants are still well available (and supported) anyway.